plerup/espsoftwareserial #237: why ttl

shownb, 2022-05-04:


dok-net, 2022-05-05:
Please provide an MCVE - all the initialization code is missing in your snippets.

shownb, 2022-05-05:
> Please provide an MCVE - all the initialization code is missing in your snippets.

sorry for my bad English. espsoftwareserial works like a charm.  i think it may be rs232_ttl converter problem.
if i run this code in this situation(esp32-->ttl-->usb-->pc),it works. pc shows that the first byte's parity is Mark.
if i run this code in another situation(esp32-->ttl-->rs232(male)<--rs232(female)-->usb-->pc). pc shows that the first byte's parity is Space. all datas are right.
```c++
#include "SoftwareSerial.h"

#define MYPORT_TX 12
#define MYPORT_RX 13

SoftwareSerial myPort;

void setup() {
  delay(2000);
  Serial.begin(115200);
  myPort.begin(19200, SWSERIAL_8S1, MYPORT_RX, MYPORT_TX, false);
  myPort.enableIntTx(true);
  if (!myPort) {
    Serial.println("Invalid SoftwareSerial pin configuration, check config");
    while (1) { // Don't continue with invalid configuration
      delay (1000);
    }
    Serial.println(PSTR("\ndone"));
  }
}

void loop() {
  byte buf[4] = { 0x11, 0x22, 0x33, 0x44};
  myPort.write((uint8_t)buf[0], SWSERIAL_PARITY_MARK);
  for (int i = 1; i < 4; ++i) {
    myPort.write((uint8_t)buf[i]);
  }
  Serial.println(PSTR("send done"));
  delay(1000);
}
```
and i also send []byte with first byte(parity is Mark) from pc  in this situation(pc-->usb-->rs232(male)<--rs232(female)-->ttl-->esp32)
esp32 show that,first byte's parity is mark.
```c++
void loop() {
  int avail = myPort.available();
  for (int i = 0; i < avail; ++i)
  {

    byte ch = (byte)myPort.read();
    Serial.print(ch, HEX);
    if (myPort.readParity()) {
      Serial.print("[m]");
    }
  }
}
```

until now,i think actually espsoftwareserial works perfect. I just have a glimmer of hope, there are someone know that, why rs232_ttl converter can not covert ttl to rs232 signal correctly. but can covert rs232 to ttl correctly.

dok-net, 2022-05-07:
As you said, this is not an issue with the library.

closed 2022-05-07 by dok-net