plerup/espsoftwareserial #238: Read() data is processed with a delay

Fantonrko, 2022-05-06:
Hello!
I am using esp32 and your library.
I don't quite understand why the following happens:

I send data to another device, the other device responds to me, I write this data to RX_buffer[] using ReadBytes() or ReadBytesUntil().
My code looks something like this:
```
testSerial.setTimeout(0);
loop(){
testSerial.write(TX_buffer[0], SWSERIAL_PARITY_MARK); // send 0x33
testSerial.write(TX_buffer[1], SWSERIAL_PARITY_MARK); // send 0x33
testSerial.readBytes(RX_buffer, 64);
Serial.println(RX_buffer[0]);                                               // need to see 0x80
Serial.println(RX_buffer[1]);                                               // need to see 0x00
testSerial.write(TX_buffer[0], SWSERIAL_PARITY_MARK); // send 0x36
testSerial.write(TX_buffer[1], SWSERIAL_PARITY_MARK); // send 0x36
testSerial.readBytes(RX_buffer, 64);
Serial.println(RX_buffer[0]);                                               // need to see 0x09
Serial.println(RX_buffer[1]);                                               // need to see 0x09
}
```

But in the serial port, when I start the program, on the first reading I see 0x00 and 0x00, and on the second reading I see 0x80 and 0x00.
On the next iteration of the loop, I see 0x09 and 0x09, and then 0x80 and 0x00.

It feels like the data is being processed only at the next iteration, but not at the moment
I hope I explained my problem clearly :)

dok-net, 2022-05-07:
Always post MCVEs, please.
Anyway, what makes you believe that `setTimeout(0)` is the correct use in your case? It makes `readBytes` return immediately no matter what. Also, your code doesn't inspect the return value from `readBytes`, so there may just as well be no data read at all into RX_buffer.
As far as I am concerned, the issue tracking in GitHub is for issues with the library, not for user questions, there are other sites better suited for peer-to-peer help, please use these first for general questions.
I hope my hints help you fix your sketch, tho :-)

closed 2022-05-07 by dok-net

Fantonrko, 2022-05-07:
Good morning, sorry for my importunity, but could you please explain reading the data in more detail?
I'm running at 9600. I'm reading data every 35 milliseconds, immediately after sending my command via write.
On the logic analyzer, I see that the data was sent successfully and I received a response. But I immediately output this answer to the serial port, and it does not show the data that I receive on the logic analyzer.

Maybe I'm not using the read function correctly. Could you show the correct application / Give material on this topic?

Fantonrko, 2022-05-09:
> Good morning, sorry for my importunity, but could you please explain reading the data in more detail? I'm running at 9600. I'm reading data every 35 milliseconds, immediately after sending my [...]

Good morning, sorry for my importunity, but could you please explain reading the data in more detail?
I'm running at 9600. I'm reading data every 35 milliseconds, immediately after sending my command via write.
On the logic analyzer, I see that the data was sent successfully and I received a response. But I immediately output this answer to the serial port, and it does not show the data that I receive on the logic analyzer.

Maybe I'm not using the read function correctly. Could you show the correct application / Give material on this topic?