plerup/espsoftwareserial #240: ESP32C3 and valid pins

fanfanlatulipe26, 2022-06-07:
Version 6.16.1
In case of ESP32C3,  only  the pin 0-1, 3 to 7, and 18 to 21 are accepted as valid in the "begin".
The 3 "strapping pins"   gpio 2, 8 and 9 are excluded.
These 3 pins can be used with HardwareSerial without restriction. (I kow we must be carefull with gpio 9 used for flashmode selection)

There is a nice ESP32C3 board available known as "LILYGO® TTGO T-01C3 ESP32-C3 " with the same form factor as the ESP8266  ESP01 module, pin compatible: the only pins available are  gpio 2, 8, 9 and RX0 (gpio20) and TX0 (gpio 21) . 

I use this module in an application where I need 3 uarts: one output only for debug, (hardware Serial0, pin 21), one for a GPS (hardware Serial1 on 8/9, and need a third Serial fora GSM module (softwareserial as ESP32C3 as only 2 hardware uarts) on 2/20

The first tests I made are OK  and it seems that we can use the pins 2/8/9 for SoftwareSerial if we change the selection of valid pins for ESP32C3:
```
In the code for bool SoftwareSerial::isValidGPIOpin(int8_t pin) {
...
...
instead of 
     return (pin >= 0 && pin <= 1) || (pin >= 3 && pin <= 7) || (pin >= 18 && pin <= 21);
change to:
      return (pin >= 0 && pin <= 9) || (pin >= 18 && pin <= 21); 
```
Regards

dok-net, 2022-06-09:
Opinions on this vary, the current checks are based on requests to make the use of the lib "safe".
I am sorry but I won't flip-flop on this issue ;-) Please consider patching your copy for your particular use case.

closed 2022-06-09 by dok-net

fanfanlatulipe26, 2022-06-09:
Thanks for the answer. I don't agree but well.....
For me I would prefer a strong warning in the documentation. Just let the user choose. Freedom  !!
And the HardwareSerial library is not so strict.

wukongxuetang, 2023-01-13:
@fanfanlatulipe26 Thanks for your  good idea！