plerup/espsoftwareserial #241: SoftwareSerial does not receive any data when running on core 1 if not initialized in core task. Send is ok.

AcuarioCat, 2022-06-13:
If SoftwareSerial is initialized (begin called) in setup and then used in a task on core 1 it sends correctly but does not receive - no data is available.
Initializing in the core task allows it to function correctly (send and receive)

Using ESP-32 WROOM-32E

Sketch below:

Using a modified version of sdm_simple.ino from the SDM library example with SoftwareSerial.

#include <SDM.h>                                                                //import SDM library

#if defined ( USE_HARDWARESERIAL )                                              //for HWSERIAL

#if defined ( ESP8266 )                                                         //for ESP8266
SDM sdm(Serial1, SDM_UART_BAUD, 12, SERIAL_8N1);                                  //config SDM
#elif defined ( ESP32 )                                                         //for ESP32
SDM sdm(Serial1, SDM_UART_BAUD, 17, SERIAL_8N1, 26, 25, false);          //config SDM
#else                                                                           //for AVR
SDM sdm(Serial1, SDM_UART_BAUD, NOT_A_PIN);                                              //config SDM on Serial1 (if available!)
#endif

#else                                                                           //for SWSERIAL

#include <SoftwareSerial.h>                                                     //import SoftwareSerial library
#if defined ( ESP8266 ) || defined ( ESP32 )                                    //for ESP
SoftwareSerial swSerSDM;                                                        //config SoftwareSerial
//SDM sdm(swSerSDM, SDM_UART_BAUD, 12, SWSERIAL_8N1, SDM_RX_PIN, SDM_TX_PIN);       //config SDM - WeMos D1
SDM sdm(swSerSDM, SDM_UART_BAUD, 17, SWSERIAL_8N1, 26, 25);       //config SDM ESP32
#else                                                                           //for AVR
SoftwareSerial swSerSDM(SDM_RX_PIN, SDM_TX_PIN);                                //config SoftwareSerial
SDM sdm(swSerSDM, SDM_UART_BAUD, NOT_A_PIN);                                             //config SDM
#endif

#endif

void readSDM() {
	Serial.print("Voltage:   ");
	Serial.print(sdm.readVal(SDM_PHASE_1_VOLTAGE), 1);            //display voltage
	Serial.println("V");

	Serial.print("Current:   ");
	Serial.print(sdm.readVal(SDM_PHASE_1_CURRENT), 1);            //display current
	Serial.println("A");

	Serial.print("Power:     ");
	Serial.print(sdm.readVal(SDM_PHASE_1_POWER), 1);              //display power
	Serial.println("W");

	Serial.print("Frequency: ");
	Serial.print(sdm.readVal(SDM_FREQUENCY), 1);                  //display frequency
	Serial.println("Hz");
}

//task on core 1
void readSDMT(void* parameter) {
        //If SDM is initialized here in the task then it both sends and receives
	sdm.begin();                             //initialize SDM communication
	for (;;) {
		readSDM();
		vTaskDelay(1000);
	}
}

void setup() {
  Serial.begin(115200);                                             //initialize serial

  // If SDM is initialized here then it will send data but not receive
  //sdm.begin();                   //initialize SDM communication

  //Read task run on core 1
  xTaskCreatePinnedToCore(
	  readSDMT,    // Function that should be called
	  "Read SDM",   // Name of the task (for debugging)
	  4096,            // Stack size (bytes)
	  NULL,            // Parameter to pass
	  2,               // Task priority
	  NULL,             // Task handle
	  1					//Core
  );
}

void loop() {
  delay(1000);                                      //wait a while before next loop
}

AcuarioCat, 2022-06-14:
I'm using the latest version 6.16.1. I'm using ESP core 2.0.3 and Visual Micro as my IDE.
All I'm trying to do is to run the process for reading from the SDM using core 1 rather than core 0.

I tried an interrupt and no, it didn't work.
So, it appears any code that is created that uses hardware interrupts has to run in the loop() and can't be run from a pinned task.

closed 2022-06-14 by AcuarioCat

dok-net, 2022-06-14:
Please read https://rntlab.com/question/esp32-designating-a-specific-core-for-specific-interrupt-service/