plerup/espsoftwareserial #242: Reading Wrong First Character

fervenceslau, 2022-06-15:
I am creating a software serial of 115200 baud rate using the serial monitor's UART0 pins 1 and 3, which works most of the time, except when reading the first byte after a reboot. The image below shows the issue I found by entering the letter 'd' multiple times. I am using a DOIT Esp32 DevKit v1 and programming with the platformio extension for vs code.

[image]

It can be seen that the first byte is not a 'd' and it can be found by taking the byte value of the input, shifting to the right by 1 position and setting the msb to 1. Ex.: 100 is 1100100 in binary and 178 is 10110010 in binary. This behavior is the same no matter the input.

The following simple code should replicate the issue shown in the attached image.

```
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <driver/gpio.h>
#include "sdkconfig.h"

#include <Arduino.h>
#include <SoftwareSerial.h>

#define SOFT_TXD    (1)
#define SOFT_RXD    (3)

SoftwareSerial mySerial(SOFT_RXD, SOFT_TXD);

void task (void *pvParameter) {
    while (1) {
        int av = mySerial.available();
        mySerial.print(av);
        if (av > 0) {
            int rd = mySerial.read();
            mySerial.print(" ");
            mySerial.print(rd);
            mySerial.print(" ");
            mySerial.print((char) rd);
        }
        mySerial.println();
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void setup () {
    pinMode(SOFT_RXD, INPUT);
    pinMode(SOFT_TXD, OUTPUT);

    mySerial.begin(115200);
    mySerial.println("\nSTARTUP");

    xTaskCreate(&task, "task", configMINIMAL_STACK_SIZE, NULL, 5, NULL);
}

void loop () {
}
```

dok-net, 2022-06-15:
I am sorry, I don't support Platformio, due to the way they handle dependencies betwenn the core and libraries, and the fact, I don't use it myself.
Additionally, if you research very recent issue discussions here, it seems that FreeRTOS tasks don't play well with the interrupt handling on ESP32. EspSoftwareSerial is intended for use with the default setup of the Arduino Core for ESP8266 and ESP32.
Finally, a bug that seems very similar to the issue you are reporting was fixed by one of the recent releases, but as I said, I am at loss to tell which version you get using Platformio :-(

dok-net, 2022-07-10:
Your sketch seems very confused, you use the same pins that are still attached to the USB to UART transceiver? Your use of a new task does help. I consider this an invalid issue.

closed 2022-07-10 by dok-net