plerup/espsoftwareserial #243: ESP32C3 Reading on pin 20 (RX0)

fanfanlatulipe26, 2022-06-23:
Version 6.16.1

Context is the same as in issue #240
There is a nice ESP32C3 board available known as "LILYGO® TTGO T-01C3 ESP32-C3 " with the same form factor as the ESP8266 ESP01 module, pin compatible: the only pins available are gpio 2, 8, 9 whitch are not accepted by ESPSerialSoftware and RX0 (gpio20) and TX0 (gpio 21) .
I use this module in an application where I need 3 uarts: one output only for debug, (hardware Serial0, pin 21), one for a GPS (hardware Serial1 on 8/9, and need a third Serial fora GSM module (softwareserial as ESP32C3 as only 2 hardware uarts) on 2/20.

With the changes I proposed in #240 I can write serial signal messages on the 4 pins 2,8,9 and 20 (verified with a logical analyzer and also with an other ESP32 reading the signals coming from the ESP32C3).

But I am unable to read any serial lines, even in a configuration when I use only the "legal" pin 20 as a receiver pin for a serial link coming from an other ESP32 generating a small message every 200ms. 
See the following test case. This test case runs prefectly with an ESP32.

```// loop on read, and  and echo on Serial
// ESP32: ok, the serial flow comming from ESP32C3 pin 2 or 8 or 9, or 20 
// ESP32C3: bad.No output. With module on programming adapter or in "stand alone" minimum configuration with a FTDI
#include <SoftwareSerial.h>
#ifdef ESP32
#ifdef CONFIG_IDF_TARGET_ESP32
#define TX0 1  // write to SerialMonitor
#define RX0 3
#pragma message "ESP32"
#elif CONFIG_IDF_TARGET_ESP32C3
#define TX0 21    // write to SerialMonitor
#define RX0 20
#pragma message "ESP32C3"
#endif
#else
#error "Need an  ESP32 or ESP32C3 !!"
#endif

SoftwareSerial SerialRX;
void setup() {
  Serial.begin(115200, SERIAL_8N1, -1, TX0); // transmit only on TX0
  
  SerialRX.begin(4800, SWSERIAL_8N1, RX0, -1); // RX only
  if (!SerialRX) {
    Serial.println("Invalid SoftwareSerial pin configuration SerialRX");
    while (1) { // Don't continue with invalid configuration
      delay (1000);
    }
  }
  delay(1000); //  forget the glitch .....
  Serial.print("Echo started ...Read on pin ");Serial.println(RX0);
}

void loop() {
  int c = SerialRX.read();
  if ( c != -1) {
  //  Serial.print(c, HEX); Serial.print(":");
    Serial.write( c);
  }
  yield();
  delay(1);
}
```

fanfanlatulipe26, 2022-06-27:
More information after some tests and debug: maybe some trouble with  ESP.getCycleCount ??
Reading a stream of "U" character (bin 0101 0101) at 38400 bds, giving a transition for each bit .
In the library, the ISR  rxBitISR fires correctly.
In rxBits(...) the number of events/transitions stored in the isrBuffer if OK. The level is OK, but it seems that the length of the transition is incorrect.
Expected number of cycles for 1 bit is m_bitCycles: 4167  (computed from ESP frequency 160Mhz and speed 38400bds)
But number of cycles for each transition, computed from the ESP.getCycleCount values stored in the isrBuffer is only 781 and so not enough for even a single bits: all the transitions are ignored.
I ran some test with ESP.getCycleCount and get very different behavior between ESP32 and ESP32C3. See following small testcase.
If ESP32C3 is waiting with a delay or a while loop the results are not the same...
2 cpu in ESP32 ??  1 cpu in ESP32C3 ??
```
// Test  ESP.getCpuFreqMHz())   ESP.getCycleCount();
// Results 
/* 
  ESP32C3:  LILYGO®TTGO T-01C3 ESP32-C3  compiled with ESP32C3 dev kit 160Mhz
14:01:39.948 -> Starting ...
14:01:39.948 -> ESP.getCpuFreqMHz(): 160
14:01:40.951 -> Nbr of cycle with while: 159852038
14:01:41.952 -> Nbr of cycle with delay: 1344803
14:01:42.953 -> Nbr of cycle with while: 159875882
14:01:43.956 -> Nbr of cycle with delay: 1339365
14:01:44.959 -> Nbr of cycle with while: 159875882
14:01:45.961 -> Nbr of cycle with delay: 1339365
Result ESP32 dev kit cpu freq 230mhz
14:08:41.546 -> ESP.getCpuFreqMHz(): 240
14:08:42.549 -> Nbr of cycles with while: 239768414
14:08:43.551 -> Nbr of cycles with delay: 239849010
14:08:44.551 -> Nbr of cycles with while: 239879752
14:08:45.553 -> Nbr of cycles with delay: 239849988

*/
void setup() {
  Serial.begin(115200);
  Serial.println("Starting ...");
  Serial.printf("ESP.getCpuFreqMHz(): %u\n", ESP.getCpuFreqMHz());
  uint32_t curCycle = ESP.getCycleCount();
}

void loop() {

  long T0 = millis();
  uint32_t startCycle = ESP.getCycleCount();
  while (millis() < T0 + 1000) {}
  uint32_t endCycle = ESP.getCycleCount();
  Serial.printf("Nbr of cycles with while: %u\n", endCycle - startCycle);
  startCycle = ESP.getCycleCount();
  delay(1000);
  endCycle = ESP.getCycleCount();
  Serial.printf("Nbr of cycles with delay: %u\n", endCycle - startCycle);
}
```

dok-net, 2022-07-10:
@fanfanlatulipe26 Could you try to prove if `micros()` is running at a stable rate?

dok-net, 2022-07-10:
The reason I am asking is because the ESP32 C3 appears to apply dynamic clock scaling for power scaling, and your measurements look like that is the case, too. It can't really be helped for RX, I am afraid, and this being so, I see no point in trying a fix for TX either.
That said, you are free to try out the SDK APIs for locking the power saving modes and see if that works - I should then probably be added to the documentation, but remain part of the user code.
If you have no time to give feedback, I am going to close this issue on short notice.

fanfanlatulipe26, 2022-07-10:
The results for micros() seem OK. I ran a test over 10s and got:
15:58:05.850 -> ESP.getCpuFreqMHz(): 160
15:58:15.825 -> Nbr of cycles with while: 29555396
15:58:15.825 -> Nbr of Usec with while: 9999937
15:58:15.825 -> 
15:58:25.845 -> Nbr of cycles with delay: 13371364
15:58:25.845 -> Nbr of Usec with delay: 9999910
15:58:25.845 -> 
15:58:35.867 -> Nbr of cycles with while: 29550261
15:58:35.867 -> Nbr of Usec with while: 10000000
15:58:35.867 -> 
15:58:45.833 -> Nbr of cycles with delay: 13371512
15:58:45.833 -> Nbr of Usec with delay: 9999916
15:58:45.833 ->
micros() use a timer instead of CCOUNT register and it may explain the results ....
I opened an issue [https://github.com/espressif/arduino-esp32/issues/6920]( https://github.com/espressif/arduino-esp32/issues/6920 ) but don't have yet feedback.
Nevertheless I have some  code running perfectly with espsoftwareserial (for RX and TX) on ESP32C3. Sure that there is no "delay()" somewhere in the loop ;-) . Will go this way for my new development on C3.
Do you think that the algorithm used in softwareserial will give good enough results if we use micros() instead of ESP.getCycleCount(), at least for slow speed ?
I will have a look also to  the power saving mode .

fanfanlatulipe26, 2022-07-14:
We can’t use the SDK APIs for locking the power saving modes. Power saving mode is not implemented in Arduino and a call to esp_pm_configure returns error ESP_ERR_NOT_SUPPORTED

Delay() is directly managed by freertos and I didn’t manage to have a look in it.

EttienneL5, 2022-07-20:
Good day,

I am having an issue where i am reading the data. When checking if there is available bytes it says that there are but when i read it, there is suddenly no data. I am using 2 esp32 wroom 32u micros. 

char * readGatewayData(size_t *returnLen) {
  char * result;

  if (swSer.available() >= 7) {
    *returnLen = swSer.available();
    result = (char*) malloc(*returnLen);
    
    int i = 0;
    while ((swSer.available() > 0) && (i < *returnLen)) {
      int x = swSer.read();
      Serial.println(x, DEC);
      if (x < 0) {
        *returnLen = 1;
        result[0] = 0xfe;
        return result;
      } else if (x > 255) {
        *returnLen = 1;
        result[0] = 0xff;
        return result;
      } else {
        result[i] = (char) x;
        i++;
      }
    }

    if (i <= 6) {
      *returnLen = 1;
      result[0] = 0xfd;
    }
  }
  return result;
}

This is my code that reads the data, the ifs are where i started to check if the value might be bigger than a byte/char

Any ideas

dok-net, 2022-07-21:
Perhaps it's just me missing dereferencing of the pointer `returnLen` in
```
returnLen = swSer.available();
```

EttienneL5, 2022-07-21:
> Perhaps it's just the missing dereferencing of the pointer `returnLen` in
> 
> ```
> returnLen = swSer.available();
> ```

I checked, i included it and i still have the issue,

So I think now after looking into it even more, Irun into the issue on other places as well. I changed it to a global array and the issue disappeared at this location but popped up at another.

I am new to pointers and yes I have an excellent tuter/dad to help me but he also could not see the issue. Now I will post the code some ware else as it has nothing to do with the library.

I am creating pointers to store "byte array data" (as if they were arrays) an then moving/passing the data in and out of functions until i know its correct. This is for my work but i am the only one that can do C in our software department so the code is not optimized or the best but it works. 
The error occurs mostly where ever i used this method. Below is a link to the project if you are interested. I can also always use some  help as i am currently behind schedule. 

Like i said, any pointers will be greatly appreciated and if I am doing stuff in the wrong way, show me.

You can never stop learning new things.

https://github.com/EttienneL5/Smart-Lock-With-Gateway.git

Thanks anyways for coming back to me.

dok-net, 2022-07-23:
I'm glad to have been able to "point" you in the right direction. I wish you success with your project.

closed 2022-07-23 by dok-net