plerup/espsoftwareserial #244: Using the listen() function will fail to receive serial port data

xiezhoubin, 2022-07-09:
[image]

xiezhoubin, 2022-07-09:
use esp32.

xiezhoubin, 2022-07-09:
sdk is 2.0.4

dok-net, 2022-07-10:
It seems your use of `listen()` in the continually running loop, just ahead of reading from the SW serial device, but resetting the RX state machine disables actually receiving any data. I don't think this use of `listen()` is intended, but nonetheless, it probably should not behave this way. Give me a minute....

dok-net, 2022-07-10:
Please try out the HEAD revision that contains a fix now.

dok-net, 2022-07-23:
I am sorry that you haven't been able to give feedback on this. Closing due to lack of activity.

closed 2022-07-23 by dok-net

xiezhoubin, 2022-07-23:
it is ok.