plerup/espsoftwareserial #246: Invert software serial signal does not pulling serial pin idle low

midelic, 2022-07-21:
I'm using a one wire communication on ES8266 and I have the  signal on idle high. Serial data  is ok but signal  is not in spec.
Normal HW Inverted signals are idle low. Any idea how to fix that?

dok-net, 2022-07-23:
Seriously, I doubt you have it all configured correctly. You haven't posted the usually required MCVE. Please make sure you are setting up the inverted mode?
```
    SoftwareSerial(int8_t rxPin, int8_t txPin = -1, bool invert = false);
    void begin(uint32_t baud, SoftwareSerialConfig config,
        int8_t rxPin, int8_t txPin, bool invert,
        int bufCapacity = 64, int isrBufCapacity = 0);
```

dok-net, 2022-08-14:
This issue will be closed soon due to lack of activity.

midelic, 2022-10-01:
This is my begin routine.I used the serial library provided with ESP8266 core. That is also yours.

swSer.begin(57600, SWSERIAL_8N1,3, 3,true, 256);
 and send data with 
    #ifdef SW_SERIAL
        swSer.flush();
        swSer.enableTx(true); //for tx
        swSer.write((uint8_t *)TxData, (size_t) sport_count);
        swSer.enableTx(false);//for rx
    #endif

dok-net, 2022-10-02:
Isn't that what you would expect to happen when calling `swSer.enableTx(false);`?

midelic, 2022-10-02:
No both RX and TX signals are inverted so I expect to be idle low not high.

midelic, 2022-10-04:
Thanks!Clear.

dok-net, 2022-11-15:
@midelic If the issue is solved / invalid, could you please close it? Thank you.

dok-net, 2022-12-14:
Assuming the issue is resolved.

closed 2022-12-14 by dok-net