plerup/espsoftwareserial #248: Add 9 bit code example?

dagnall53, 2022-08-01:
Would it be possible to add a simple 9 bit code example please.
I am having trouble following how you suggest the Mark and Space can be used in practice - But I suspect the answer is very simple and I just have not "seen" how it can be done.
Many thanks

dok-net, 2022-08-14:
@fredvik It would be wonderful if you could help out here with a canonical example?

fredvik, 2022-08-15:
Wow, it suddenly feels like a very long time ago since I was active in this project. Time flies :-)

I haven't coded much since then, so the code I have available is based on an obsolete and patched version of espsoftwareserial and is probably of questionable quality as I'm not a very skilled programmer. Also, the library has developed since I discontinued my project so I don't have any current code that is even nearly functional. 

My use case was that I needed to decode a protocol used for heat pump monitoring and control over RS484. It was setup as 8 bit with 1 parity bit used as MARK or SPACE based on context.

Each transaction is intiated from the heatpump and starts with a packet where the first two bytes are the address of the intended receiver (hardcoded to 0x00 0x14), both bytes sent with MARK parity. The receiver then responds with ACK (0x06) to confirm it is ready to listen.

The second (content) packet begins with the sender address (1 byte), the length of the packet (1 byte), the message content (what I call the "telegram") and it all ends with a checksum (1 bytes). The message content (telegram) is of varying length and depending on content information is sent as pairs of bytes. Upon reciept, the receiver confirms with an ACK if the checksum is correct. Alternatively, the receiver can send NAK (0x15) if the computed checksum did not agree with the receive done. The heatpump then closes the transaction by sending ETX (0x03) with MARK parity. After waiting a while the heat pump will start a new transaction by sending a packet beginning with two address bytes in MARK parity.

This type of protocol using MARK parity for addressing is not too uncommon, especially in older protocols. The general idea is that a receiver only needs to listen to addresses when the parity is MARK ([node address flagging](https://stackoverflow.com/questions/13953095/what-is-the-difference-between-using-mark-space-parity-and-parity-none)). If it is not the receiver's own address it can ignore the communication and save its power until another address is received. This can be easily implemented in hardware so that the UART only triggers the main processor when bytes with MARK parity are received. That way a node can stay idle most of the time even if lots of packets are sent on the bus.

In software, we need to listen to each character, check if it has MARK parity and treat it accordingly.

My [code is here](https://github.com/fredvik/NIBEBridge/blob/master/src/NIBEConnect.cpp). The loop first reads the parity and the received byte as _inpar_ and _inbyte_ (line 26-27). Then there is a state machine (the case statments) to keep track of the protocol/communication to know when to read and when to ignore communication on the bus.

Maybe not the best code example but this is what I have. I'm happy to respond to any questions but since I gave up on my project 4 years ago I am no longer up to speed with my code and I have not kept pace with the implementation in espsoftwareserial.

dagnall53, 2022-09-27:
@fredvik  The link you provided seems to have disappeared?  Would it be possible for you to reinstate it? 
Cheers and thanks

fredvik, 2022-09-27:
The repository was set to _private_, have changed now it to public. 
Sorry for the inconvenience!

dagnall53, 2022-09-27:
Many thanks.. I will look through it in detail later.
I'm actually exploring options for some UART code that broke with the 2.0.2
(+) compiler changes that changed how the interrupts are coded.
My previous code attached / detached the edge triggered interrupt inside a
timer interrupt. (worked fine up until the 2.0.2 changes, and then it
crashed.)
Something to do with changing to a (better?) esp32 compatibility, but I do
not understand it!
I've got a version of my code that uses a flag now to avoid the crash, but
was looking at the existing SW Serial UARTS and I see they seem to use just
the edge triggered interrupt and then do timing for the bits without using
the timing interrupt..  As I'm not really a Object oriented C code guy, Ive
been finding it difficult to follow the examples etc.. So many thanks for
the example..
Any insights into why the compiler needed changing at 2.0.2 (if you know)
would be helpful!..

dok-net, 2022-12-14:
This issue is stale.

closed 2022-12-14 by dok-net