plerup/espsoftwareserial #249: SoftwareSerial GPIO2 as RX not working

bmitov, 2022-08-11:
I am trying to read SW Serial from GPIO2 . It does not appear to work :-(

#include <SoftwareSerial.h>

#define rxPin 2
#define txPin 13

// Set up a new SoftwareSerial object
SoftwareSerial mySerial(rxPin, txPin);

void setup()  {
    // Define pin modes for TX and RX
    pinMode(rxPin, INPUT);
    pinMode(txPin, OUTPUT);
    
    // Set the baud rate for the SoftwareSerial object
    mySerial.begin( 9600 );
    Serial.begin( 9600 );
}

void loop() {
    if (mySerial.available() > 0) {
        Serial.write( mySerial.read() );
    }
}

GPIO4 as example work, unfortunately the hardware that I use requires the Serial to be on GPIO2 :-( .
Is this a bug? Any suggestions?

dok-net, 2022-08-14:
This has been answered recently. Please remember to research closed issues, too, before posting requests for help here. Good luck.

closed 2022-08-14 by dok-net

michaelosthege, 2024-05-20:
Looks like the previous comment may have referred to https://github.com/plerup/espsoftwareserial/issues/261 (?)

Essentially GPIO2 is considered as an invalid pin by this library.