plerup/espsoftwareserial #251: add "weak" attribute to function isValidGPIO , isValidRxGPIO , isValidTxGPIO

fanfanlatulipe26, 2022-08-18:
This request is a followup of #240  (valid pins for RX/TX, strapping pins ...)
Please could you add  "weak" attribute to the functions:
```
bool isValidGPIOpin(int8_t pin);
bool isValidRxGPIOpin(int8_t pin);
bool isValidTxGPIOpin(int8_t pin);

bool isValidGPIOpin(int8_t pin)  __attribute__ ((weak)); 
bool isValidRxGPIOpin(int8_t pin) __attribute__ ((weak));
bool isValidTxGPIOpin(int8_t pin) __attribute__ ((weak));
```

In this way the user can replace them and still use the standard library.
No need for "patched" local version of the library. No compatibility issues.
Regards

dok-net, 2022-10-02:
Is there any particual reason you prefer C-style weak linkage over C++ polymorphism?

fanfanlatulipe26, 2022-10-02:
Sorry, I am not a C++ expert (nor even a C one  !) ... Just trying to play a bit with microcontrolers...
 I just found that it was  very easy to use my own isValidGPIOpin if the function is declared with the weak attribute in the library. 
Just need to write my own function and the linker will gives it priority.
I don't know how it can be done in the C++ polymorphism context.

dok-net, 2022-12-14:
Keeping #261, closing this one.

closed 2022-12-14 by dok-net