plerup/espsoftwareserial #252: Compile errors when changing build flags.

JimDrewGH, 2022-09-14:
Recently, there was a change to the TFT_eSPI library to get around the inconvenience of the Arduino IDE not allowing sketch #defines to be visible to a library.  This work around uses a file placed in the board hardware package's root folder. ie:

c:\Users\user_name\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\

The file is called platform.local.txt and is part of the compiler system, designed to allow changes to the compiler flags.  The file contains this line:

build.extra_flags=-I{build.source.path}

When this is used, there are literally hundreds of compiler errors with the espsoftwareserial library.  No other library (so far) has had this problem.

Can you look into this issue to determine why it is occurring?

For reference, there is a discussion here:

https://github.com/Bodmer/TFT_eSPI/discussions/1969

Thanks!

dok-net, 2022-10-02:
I'm sorry, I don't have the resources to track down your issue. Please report here once you find the core problem.

dok-net, 2023-01-22:
Closing due to lack of activity.

closed 2023-01-22 by dok-net