plerup/espsoftwareserial #254: Extra byte sent on the serial

cguimaraes, 2022-09-16:
While testing the serial communication between an ESP32 and a laptop, it seems that the ESP32 is setting an extra byte before the buffer to be sent.

ESP32 running Arduino and using EspSoftwareSerial.
Laptop running macos/linux and using https://github.com/ZettaScaleLabs/z-serial

The https://github.com/ZettaScaleLabs/z-serial was already validated for laptop to laptop serial communication as well as for laptop to microcontroller running MBedOS. This makes me assume that the issue might be introduced by this library.

Both serials are configured with a baudrate of 115200 and SWSERIAL_801.
COBS is being applied so that the buffer to be sent does not have zeros, except for the last one that notifies the last byte.

ESP-32 logs:
```
Buffer with bytes to be sent: [02 0C 11 03 06 04 08 A1 AB 22 6F FE 4D 69 53 2D 16 B6 F9 00]
# of bytes written in the serial (return value from write):20 bytes
```

Laptop logs:
```
Read 21 bytes. 
Buffer with received bytes: [80, 02, 0C, 11, 03, 06, 04, 08, A1, AB, 22, 6F, FE, 4D, 69, 53, 2D, 16, B6, F9, 00]
```

As you can see 0x80 (the first received byte in the laptop was not actually sent by the ESP32). No matter how many times I restart and/or connect the Serial and the applications that value does not change.

Any clue on what might be happening?

DaanTheoden, 2022-09-16:
Hello,

Did you try any of the other configurations? 

```
enum SoftwareSerialConfig {
    SWSERIAL_5N1 = SWSERIAL_PARITY_NONE,
    SWSERIAL_6N1,
    SWSERIAL_7N1,
    SWSERIAL_8N1,
    SWSERIAL_5E1 = SWSERIAL_PARITY_EVEN,
    SWSERIAL_6E1,
    SWSERIAL_7E1,
    SWSERIAL_8E1,
    SWSERIAL_5O1 = SWSERIAL_PARITY_ODD,
    SWSERIAL_6O1,
    SWSERIAL_7O1,
    SWSERIAL_8O1,
    SWSERIAL_5M1 = SWSERIAL_PARITY_MARK,
    SWSERIAL_6M1,
    SWSERIAL_7M1,
    SWSERIAL_8M1,
    SWSERIAL_5S1 = SWSERIAL_PARITY_SPACE,
    SWSERIAL_6S1,
    SWSERIAL_7S1,
    SWSERIAL_8S1,
    SWSERIAL_5N2 = 0200 | SWSERIAL_PARITY_NONE,
    SWSERIAL_6N2,
    SWSERIAL_7N2,
    SWSERIAL_8N2,
    SWSERIAL_5E2 = 0200 | SWSERIAL_PARITY_EVEN,
    SWSERIAL_6E2,
    SWSERIAL_7E2,
    SWSERIAL_8E2,
    SWSERIAL_5O2 = 0200 | SWSERIAL_PARITY_ODD,
    SWSERIAL_6O2,
    SWSERIAL_7O2,
    SWSERIAL_8O2,
    SWSERIAL_5M2 = 0200 | SWSERIAL_PARITY_MARK,
    SWSERIAL_6M2,
    SWSERIAL_7M2,
    SWSERIAL_8M2,
    SWSERIAL_5S2 = 0200 | SWSERIAL_PARITY_SPACE,
    SWSERIAL_6S2,
    SWSERIAL_7S2,
    SWSERIAL_8S2,
};
```

I would suggest trying without parity: `SWSERIAL_8N1`.

Best regards,

Daan

cguimaraes, 2022-09-16:
I tried with different configurations, including different data bits, parity, stop bits and also baudrate.
The result is always the same: an extra initial byte.

In fact, it is always the first byte sent in the serial after the reboot of the ESP32.
All the subsequent messages seem to be properly sent afterwards.

Still, this extra byte is kind of annoying to be handled on the other side.

JimDrewGH, 2022-09-17:
Are you sure the problem is the 1st byte sent and not the 1st byte received?  This is a common issue with microcontrollers in general when powering up at different times.   If the 1st device powers up and expects the other device to already be online (when it's not and the serial line is low because it is not being driven high yet) THAT will cause a START bit condition to occur and of course all low's is 0x00.  You should always read the serial port's FIFO until you get an error (no more data) as part of the initialization.

cguimaraes, 2022-09-17:
@JimDrewGH in fact I believe that it is what is happening.
I was just not expecting that to be always the same byte (and the fact that with other microcontrollers under the same conditions that was not happening), so I was wondering if it could any kind of issue created by the library.

I am already following your suggestion:
 1. Upon initialization I am forcing the serial to send 0x00.
 2. On the receiver side, it will be detected as an invalid COBS ( [0x80 0x00] ), thus dropping whatever was received before.
 3. Communications can be established normally.

Thanks for the advice.

DaanTheoden, 2022-09-17:
If it always reproduces in this exact way, what would be the issue with a read routine that filters out every 0x80 byte when the receive buffer size < 2?

JimDrewGH, 2022-09-17:
You don't want to be SENDING anything on power up.  You want to ONLY clear the FIFOs on both ends.  You can do that with a typical:

```
  while (Serial.available()) {                  // if serial data is present
    Serial.read();                              // capture it so it doesn't appear
  }

```

If you don't clear the FIFO on the receiver side you are ALWAYS going to be getting the invalid first byte.

dok-net, 2022-10-02:
Should this issue be closed as solved?

cguimaraes, 2022-10-02:
Since it seems that it is not a direct issue from the library but from the code using it, I would say we can close it.

Thank you @JimDrewGH and @DaanTheoden for helping and sharing your knowledge.

closed 2022-10-02 by cguimaraes