plerup/espsoftwareserial #257: REGRESSION: 6.16.1 no longer works at 9600 8N1 on ESP32S2, 6.15.2 works on same hardware/project

avillacis, 2022-09-30:
Hardware: ESP32S2 custom board, with TX on pin 2, RX on pin 1, RS485 data-enable on pin 3.

With latest version of EspSoftwareSerial (6.16.1), our Modbus/RS485 project can no longer communicat with our Modbus devices at 9600 baud 8N1. The exact same project still works correctly when EspSoftwareSerial is manually reverted to version 6.15.2.

By far the most likely cause is https://github.com/plerup/espsoftwareserial/pull/231 which might accelerate pin use under ESP8266 but makes no mention of having been tested under ESP32S2.

ESP32 has not been checked for regression but might be affected too.

Workaround is, as mentioned, manual revert to 6.15.2.

dok-net, 2022-10-02:
Is it true that your issue doesn't strictly relate to the bps, but rather ESP32-S.. or -C... not supporting the latest versions of ESPSoftwareSerial? First, keep in mind that EspSoftwareSerial first and foremost is for the ESP8266 MCUs. It's also (mostly) always vetted for the ESP32 by my. I have gathered from recent issues here, that there is a frequency scaling in the later ESP32-xxx MCUs that breaks the timing algorithm of EspSoftwareSerial. I don't have the resources to address this in the near future. Consider these later ESP32-family MCUs unsupported out-of-the-box for that reason. You may be able to circumvent your software serial issues by locking the MCU processor to a single frequency from your own code. If you are sucessful, I am looking forward to PR for integrating such locks into EspSoftwareSerial - again, it may take a bit of time for me to review, comment, and eventually merge any such changes.
EspSoftwareSerial is NOT unsupported, but it does what it is supposed to do, viz. run on the ESP8266 and plain ESP32.

dok-net, 2022-10-02:
@avillacis On second thought, THIS issue is a duplicate of #255, isn't it? The problem is not with the library, but the frequency scaling on the later ESP32-family MCUs from Espressif.
You might want to close THIS issue.

avillacis, 2022-10-03:
> I have gathered from recent issues here, that there is a frequency scaling in the later ESP32-xxx MCUs that breaks the timing algorithm of EspSoftwareSerial.

Hmmm... this does not explain why the 6.15.2 version does work correctly with the ESP32-S2 in my case. In case it matters, I always use Arduino IDE, with CPU frequency at 240 MHz, since I have no reason to use a lower frequency. In fact, this is the first time I hear of a supposed issue dynamic frequency breaking something.

> I don't have the resources to address this in the near future. Consider these later ESP32-family MCUs unsupported out-of-the-box for that reason. You may be able to circumvent your software serial [...]

I might be able to check which commit of the pull request (https://github.com/plerup/espsoftwareserial/pull/231) was the one that broke ESP32-S2, but it will take me some time, as I am busy with my day job. I am more willing to believe that the pin register map manipulation is the one at fault here, and not necessarily an issue with dynamic frequency.

avillacis, 2022-10-03:
> @avillacis On second thought, THIS issue is a duplicate of #255, isn't it? The problem is not with the library, but the frequency scaling on the later ESP32-family MCUs from Espressif. You might want to close THIS issue.

As I mentioned above, the frequency scaling theory does not explain why version 6.15.2 of this library keeps working, on the same ESP32-S2, using the same Arduino-ESP32 2.0.5 as before, and using the same default of 240MHz I always do. As such, I will keep this issue open while I report back with my own tests.

avillacis, 2022-10-03:
Another data point. I was requested off-list to run the demo sketch shown at [this bug report](https://github.com/espressif/arduino-esp32/issues/6920) on my ESP32-S2. The result ([here](https://github.com/espressif/arduino-esp32/issues/6920#issuecomment-1266039862)) show there appears to be no issue affecting the `delay()` call on the ESP32-S2.

dok-net, 2022-11-12:
@avillacis You mentioned that your's is a custom board. Could you please check that the `portInputRegister()`, `digitalPinToBitMask()`, and `portOutputRegister()` functions are correct for your hardware?

avillacis, 2022-11-13:
> `portInputRegister`

I am using the "ESP32S2 Dev Board" Arduino configuration with my board. The board is "custom" only in the sense that it is a custom-made PCB designed to provide some peripheral access around a standard ESP32-S2 WROVER. Our projects do not change the standard Arduino-ESP32 definitions for the functions you mention. Additionally there have been no problems with GPIO manipulations with other projects using our board, so I do not really understand why these functions would not be working properly on the board.

Just tested the `ccystous` branch and it appears to work correctly. https://github.com/plerup/espsoftwareserial/issues/255#issuecomment-1312807212

avillacis, 2022-11-13:
While setting up my environment for a git bisect to find out which commit introduced the regression, I found out I cannot longer reproduce the original bug, even with version 6.16.1. I am a bit worried about issues like this that disappear when I try to to get a closer look at them.

For the moment, the bug does not appear any longer, so I am closing this report.

closed 2022-11-13 by avillacis