plerup/espsoftwareserial #258: Cleanup compile warning

pabloandresm, 2022-11-05:
Hello there,

Compiling with extra warning flags I've noticed an annoying warning (that even brakes compilation if -Werror).

........\libraries\EspSoftwareSerial\src\SoftwareSerial.cpp: In member function 'virtual size_t SoftwareSerial::readBytes(uint8_t*, size_t)':
........\libraries\EspSoftwareSerial\src\SoftwareSerial.cpp:296:38: warning: suggest braces around empty body in an 'else' statement [-Wempty-body]
         else optimistic_yield(1000UL);

The warning is self explanatory. Just braces around the else statement should fix this.

Thanks

dok-net, 2022-11-06:
Thanks for pointing this out. I probably only occurs on targets where optimistic_yield is an empty define. The fix will be in a future release.

closed 2022-11-06 by dok-net

pabloandresm, 2022-11-06:
If you want I can generate a pull request.