plerup/espsoftwareserial #261: Use GPIO2 as TX

marcootaviomello, 2022-11-29:
I need to use the GPIO2 as a TX, I know that it is a stap pin, but a PCB version has already been produced considering it as a TX, by default I know that the lib does not allow use, can someone tell me in the source code where I should comment to remove this restriction?

fanfanlatulipe26, 2022-11-30:
Have a look at #251 
The test is in bool isValidGPIOpin(int8_t pin);

dok-net, 2022-12-14:
The library will be not reverted to allow the use of official strapping pins, this has been stated in prior issues to the same effect here.
The issue can be left pending as a reminder, that perhaps some mechanism to override this restrictive tests could be implemented in a future release.

hemchik, 2023-02-09:
Здравствуйте, как с вами связаться?

dok-net, 2023-02-18:
I am working on implementing this via template syntax. Bear with me.

dok-net, 2023-02-18:
@mcspr any opinion on this from the ESP8266 team?

dok-net, 2023-03-05:
In master, there is now the BasicUART class template. By providing a custom implementation for GpioCapabilities, one case permit or restrict the use of any GPIOs, strapping pin or not, etc.

dok-net, 2023-03-05:
@fanfanlatulipe26 Hi, I hope you will be glad to hear that your request has now been resolved via C++ template syntax. Here's a little advice on what needs to be done, if you are not completely privy to C++:

The default `SoftwareSerial` class is now a template class, aliased from `BasicUART< GpioCapabilities >`. Your mission, if you accept it, is to provide a duck typed implementation of `IGpioCapabilities`, possibly with no checks at all that the given GPIOs conform to the requested capabilities.
Say you name that instance class `UncheckedGpioCapabilities`. You would do that in your code, no need to change the library sources. Then do this:
`using PromiscuousUART = BasicUART< UncheckedGpioCapabilities >;`
And there you go, your new `PromiscuousUART` will allow you to configure it for any GPIO you wish.

dok-net, 2023-03-06:
@fanfanlatulipe26 you might ;-) I hope you'll be able to figure out how to copy, paste and modify to get where you need to be, given my hints from earlier. I'm closing this issue because it's resolved.

closed 2023-03-06 by dok-net