plerup/espsoftwareserial #262: REGRESSION: data write at 9600 8N1 plus enableIntTx(false) is unreliable since 6.16.0 (fix identified)

avillacis, 2022-12-05:
This bug report is a continuation of https://github.com/plerup/espsoftwareserial/issues/257 closed because of insufficient information on how to reproduce.

As before, hardware is ESP32-S2 running a Modbus/RS485 project using this library. Half-duplex communication is achieved by invoking `enableIntTx(false)` on the SoftwareSerial object. Communication runs at 9600 baud 8N1. Data interchange is reliable with library version 6.15.2 but is intermittently corrupted in 6.16.0 and later.

The root cause of this issue is a change of behavior introduced when refactoring the `preciseDelay()` into two different methods. A busy-loop inside the lazy delay scenario was previously executed OUTSIDE the critical section, but the refactor moved this particular busy-loop INSIDE the critical section. By moving this busy-loop back outside the critical section, I was able to fix the issue, in both 6.16.1 and recently-released 6.17.0.

avillacis, 2022-12-05:
I have created the pull request containing the fix: https://github.com/plerup/espsoftwareserial/pull/263

avillacis, 2022-12-05:
> @avillacis Thanks for reporting this regression. You haven't quite correctly identified what the critical section, as you call it, is, therefore I believe your fix is incorrect. In any case, now that I had to take another look, it's a sorry mistake with unsigned arithmetic, that I am surprised hasn't been showing in my tests much more frequently. Could you please test that my attached PR fixed this issue for you? Thanks.

NAK. The referenced pull request does NOT fix the intermittent write corruption I experience. Just tested on the exact same project on the exact same hardware.

dok-net, 2022-12-05:
@avillacis Could you please check and correct your use of "critical section", as I am certain you have that backward?
Running `preciseDelay` with interrupts disabled, according to the way you describe the malfunction, would not properly assure that `lazyDelay` does not end early. But that would imply that `preciseDelay` never properly works outside `lazyDelay` either. Which isn't the case in any of my tests. I am not running on an ESP32-S2 to be sure, though.

dok-net, 2022-12-05:
@avillacis Could you please check and correct your use of "critical section", as I am certain you have that backward?
Running preciseDelay with interrupts disabled, according to the way you describe the malfunction, would appear to not properly assure that lazyDelay does not end early. But that would imply that preciseDelay never even properly works outside lazyDelay either. Which isn't the case in any of my tests. I am not running on an ESP32-S2 to be sure, though.

avillacis, 2022-12-05:
> 

I am not sure I understand what you are saying I have "backwards".

From what I understand of the code, the `::write()` method, when interrupts-on-tx is disabled, disables interrupts, which in the ESP32 platform is done by entering the critical section. Inside, it then calls `::writePeriod()`, which in turn chooses whether to call `::preciseDelay()` or `::lazyDelay()`. When choosing `::lazyDelay()`, the code starts by enabling interrupts, which in the ESP32 platform, is done by (temporarily) exiting the critical section. It then does the coarse delay, then disables interrupts again (on ESP32, enters the critical section again). My pull request just moves the `::preciseDelay()` call nested inside `::lazyDelay()` from inside the re-entered critical section, to just outside it. The equivalent behavior was to have the busy-loop before the interrupt-disabling at the end of the lazy-loop scenario, and my pull request just restores this previous (working) behavior.

I was pointed to the particular commit by a bisection with a recompile of my sketch at each step. My pull request is the only change on top of 6.17.0, and fixes the write in my case. I do not have any other changes on the critical section itself.

Just in case, I tested removing the nested `::preciseDelay()` call altogether from the body of `::lazyDelay()`, and it results in what appears to be a read corruption of the Modbus response, specifically the first byte identifying the slave device that answered.

dok-net, 2022-12-05:
> I am not sure I understand what you are saying I have "backwards".
I am sorry, I got confused about that myself, actually.

avillacis, 2022-12-05:
> A question on the side: Is your use of EspSoftwareSerial for Modbus a closed-source endeavor? If it's not, I might be interested in testing on the ESP32-C3 myself.

Unfortunately, it is closed-source, and not publicly available.

However, I have just made a sample sketch using both ModbusMaster and EspSoftwareSerial, to exercise the bug. This sketch reliably timeouts with current 6.17.0 and reliably reads correctly with my proposed pull request. The ModbusMaster library the one at https://github.com/4-20ma/ModbusMaster , available through Arduino IDE, at version 2.0.1.

```C++
#include <Arduino.h>
#include <ModbusMaster.h>
#include <SoftwareSerial.h>

// Change these pin definitions as required
#define RS485_SERIAL_RX GPIO_NUM_1
#define RS485_SERIAL_TX GPIO_NUM_2
#define RS485_DATA_EN GPIO_NUM_3

// Board-specific relay control to enable 5V, should be optional if
// proper energy is provided to RS485 in some other way
#define ENABLE_5V GPIO_NUM_4

#define RS485_SERIAL ESP_Serial
#define RS485_MODBUS_BAUD 9600

// Change as required for test device
#define MODBUS_SLAVE_ADDR 0x01
#define MODBUS_ADDR_DATA 0x06
#define MODBUS_LEN_DATA 1

SoftwareSerial ESP_Serial;
ModbusMaster mbus;

void RS485_switch2TX(void) {
  digitalWrite(RS485_DATA_EN, HIGH);
}

void RS485_switch2RX(void) {
  digitalWrite(RS485_DATA_EN, LOW);
}

void setup() {
  delay(1000);
  Serial.begin(115200);

#ifdef ENABLE_5V
  pinMode(ENABLE_5V, OUTPUT);
  digitalWrite(ENABLE_5V, HIGH);
#endif

  Serial.println("RS485 setup...");

  pinMode(RS485_DATA_EN, OUTPUT);
  RS485_SERIAL.begin(RS485_MODBUS_BAUD, SWSERIAL_8N1, RS485_SERIAL_RX, RS485_SERIAL_TX, false);
  RS485_SERIAL.enableIntTx(false);

  Serial.println("Modbus setup...");
  mbus.begin(MODBUS_SLAVE_ADDR, RS485_SERIAL);
  mbus.preTransmission(RS485_switch2TX);
  mbus.postTransmission(RS485_switch2RX);

  // Delay for slave device startup
  Serial.println("Waiting for device startup...");
  delay(5000);
}

void loop() {
  uint8_t r = mbus.readHoldingRegisters(MODBUS_ADDR_DATA, MODBUS_LEN_DATA);
  if (r == 0) {
    Serial.printf("SUCCESS readHoldingRegisters, value read is 0x%04x\r\n", mbus.getResponseBuffer(0x00));
  } else {
    Serial.printf("FAILED readHoldingRegisters, error code 0x%02x\r\n", r);
  }

  delay(500);
}
```

This sketch has been tested on my ESP32-S2 board. I do not have any ESP32-C3 to test it with.

dok-net, 2022-12-08:
@avillacis Come to think of it, RS485, two wire by any chance?

avillacis, 2022-12-08:
> @avillacis You'll be glad to find your requested change is now part of the PR #264 . Please test against your application - thanks for reporting this, I wouldn't have seen the problem in my tests/applications, mostly on ESP8266 or dual-core ESP32.

I have just tested the pull request, and it fixes the issue for me.

> @avillacis Come to think of it, RS485, two wire by any chance?

The project uses Modbus over RS485. This uses the standard layout of 4 cables: A, B, GND, 5V.

closed 2022-12-10 by dok-net