plerup/espsoftwareserial #265: Arduino IDE error when compilig

hmandelik, 2022-12-14:
When compiling on the new Arduino IDE (2.xx) we receive an error message for the  functions _isValidGPIOpin_ _isValidTxGPIOpin_ _hasRxGPIOPullUp_ as follows: "enclosing class of constexpr non-static member function XXX is nota a literal type"

I am not very familliar with C++, anyone could help with this?

hmandelik, 2022-12-14:
Bellow the complete error message compiling the "bitpattern" sketch for a DOIT ESP32 DEVKIT V1 board:

\libraries\EspSoftwareSerial\src/SoftwareSerial.h:226:20: error: enclosing class of constexpr non-static member function 'bool SoftwareSerial::isValidGPIOpin(int8_t) const' is not a literal type
     constexpr bool isValidGPIOpin(int8_t pin) const;
                    ^
\libraries\EspSoftwareSerial\src/SoftwareSerial.h:87:7: note: 'SoftwareSerial' is not literal because:
 class SoftwareSerial : public Stream {
       ^
\libraries\EspSoftwareSerial\src/SoftwareSerial.h:87:7: note:   'SoftwareSerial' has a non-trivial destructor
\libraries\EspSoftwareSerial\src/SoftwareSerial.h:227:20: error: enclosing class of constexpr non-static member function 'bool SoftwareSerial::isValidRxGPIOpin(int8_t) const' is not a literal type
     constexpr bool isValidRxGPIOpin(int8_t pin) const;
                    ^
\libraries\EspSoftwareSerial\src/SoftwareSerial.h:228:20: error: enclosing class of constexpr non-static member function 'bool SoftwareSerial::isValidTxGPIOpin(int8_t) const' is not a literal type
     constexpr bool isValidTxGPIOpin(int8_t pin) const;
                    ^
\libraries\EspSoftwareSerial\src/SoftwareSerial.h:230:20: error: enclosing class of constexpr non-static member function 'bool SoftwareSerial::hasRxGPIOPullUp(int8_t) const' is not a literal type
     constexpr bool hasRxGPIOPullUp(int8_t pin) const;
                    ^

exit status 1

Compilation error: exit status 1

phaltness, 2022-12-14:
> When compiling on the new Arduino IDE (2.xx) we receive an error message for the functions _isValidGPIOpin_ _isValidTxGPIOpin_ _hasRxGPIOPullUp_ as follows: "enclosing class of constexpr non-static [...]

just checkout commit dead84f775695d7bec1f1b76437f217556f4ddb3 - it should work till bug will be fixed...

ali80, 2022-12-14:
thanks for the fast solution, this was working before, suddenly the error appeared, now fixed temporarily

dok-net, 2022-12-14:
@hmandelik Please allow me to remind you that entering issues without stating the version of EspSoftwareSerial you are referring to, is nearly futile.

@phaltness:
> just checkout commit https://github.com/plerup/espsoftwareserial/commit/dead84f775695d7bec1f1b76437f217556f4ddb3 - it should work till bug will be fixed...

That is an amuzeing statement, given that since the revision you mention there have been two releases 6.17.0 and 6.17.1.

dok-net, 2022-12-14:
@ali80 "temporarily"? Without very specific information on what versions of the Arduino Core for ESP32 and EspSoftwareSerial are in use, whether you have remembered to setup the toolchain if it's a Git clone (`cd tools; ./get.exe` on Windows), or if Arduino IDE 2.0 (2.0.3?) unexpectedly contains a version of of the Core in it's installation setup, this issue provides no sufficient information to work on it.
Can anyone please fill in the gaps?

I have successfully built the mentioned example sketch on a fresh setup of Arduino IDE 2.0.3 with my properly, ahead of time, configured Arduino Core for ESP32 using just the IDE's drop down menus for configuration. In my mind, it's unlikely that Arduino has done anything to provide diverging compiler toolchains from what Espressif downloads via the `get.exe` utility. Therefore, my working hypothesis is that you are dealing with inconsistent setups. Not an issue at all to track here :-)

hmandelik, 2022-12-15:
Sorry guys for the lack of info, I am using the Arduino IDE 2.03 but also tried on earlier version (1.8.18) and got the same error. The library version is 6.17.1. Thank you for all the support ;)

hmandelik, 2022-12-15:
I have revearsed the library to the 6.16.1 version and worked fine!

ali80, 2022-12-15:
@dok-net sorry for bad info, and thanks for your awesome work
as to some context, I've been using this lib for more than a year in `platformio`, using fixed version
`plerup/EspSoftwareSerial@^6.15.2`
had no problems before but suddenly I got errors without ANY change to my code.
```bash
PlatformIO Core version: 6.1.5
esp32 platform: espressif32@3.3.2
```
cleaned `platofrmio`'s lib cache and build cache, did not help,
updated to latest `EspSoftwareSerial` which I believe is `plerup/EspSoftwareSerial@^6.17.1`, also didnt help.
cloned the repo and checkout out to this commit `dead84f`, it is fixed
let me know if there is anything missing, and thanks again

PR #267 (2022-12-31): Release 7.0.0
This release contains two changes:
- Fixes #265 , the constexpr compiler errors reported on ESP32
- Drops `perform_work` in favor of HW IRQ-based callback that may be useful for power saving by suspending the task that handles SW serial input.
closed 2022-12-31 by dok-net