plerup/espsoftwareserial #268: I am able to send the data but unable to receive it.

nimish21, 2023-01-03:
Hello, Can anyone help me with the sample code to receive data correctly? Please help me to resolve this issue.

My Code:

#define RS232TX 26
#define RS232RX 25

SoftwareSerial RS232;

bool rs232Configure = false;
bool rs232Init(uint32_t baudRate){
  RS232.begin(baudRate, SWSERIAL_8N1, RS232RX, RS232TX, false, 95);
  for(uint8_t cnt=0; cnt<5; cnt++){
    RS232.print("+");
    delay(250);    
  }
  RS232.flush();
  if(RS232){ Serial.println("[  RS232  ] RS232 Port Configured Successfully! "); rs232Configure = true; }
  else { Serial.println("[  RS232  ] RS232 Port Not Configured Properly! "); rs232Configure = false; }
  return rs232Configure;
}

void rs232Send(String rs232Buffer){
  if(rs232Configure){
    RS232.flush(); RS232.enableTx(true);
    Serial.println("[  RS232  ] RS232 Buffer Sent! Buffer: "+rs232Buffer);
    RS232.print(rs232Buffer);
    RS232.enableTx(false);
  }
}

void rs232Read(void){
  String rs232Buffer;
  if(rs232Configure == true && RS232.available()){
    RS232.enableTx(false);
    while (RS232.available()) {
      char recv = RS232.read();
      rs232Buffer = rs232Buffer + String(recv);
    }
    Serial.println(rs232Buffer);
    //return rs232Buffer;
  }
}

void rs232Loop(void){
  while (RS232.available() > 0) {
    Serial.write(RS232.read());
  }

  while (Serial.available() > 0) {
    RS232.write(Serial.read());
  }
}

nicklasb, 2023-01-06:
Not much information, I am afraid.
What are you trying to communicate with?
How do you know that you are sending successfully?

svdrummer, 2023-01-07:
Where is your baud rate settings
Where is your Serial.begin()
and
RS232.begin()

Also, your loop to check the rs232Read subroutine

nimish21, 2023-01-09:
Hello Everyone, 
Let me share the main code too. I want to establish SoftSerial Communication via RS-232 (Max232 IC). I have connected an RS232 serial monitor device to the Esp32.

The connections are below:
ESP32 (TXRX) -- MAX232 -- RS232 Serial Monitor -- PC (Hercules Tool) 

Complete Code:
**[Blink.txt] (https://github.com/plerup/espsoftwareserial/files/10371054/Blink.txt)**

Comments:
@nicklasb, I have connected the serial monitor to esp32 for the debugging purpose. If you can check I am sending the '+' character 5 times after initializing (begin) the serial communication.

@svdrummer, please check line number 9. I have declared a function for serial configuration. That function has been called on line number 34.

dok-net, 2023-01-10:
Is yours an Arduino sketch, why are you omitting the exact `setup()` and `loop()` functions?
Are you certain that you have connected the MAX232 properly? Does your sketch work if you change from SW Serial to one of the ESP32's HW UARTS instead, provided you are using a "classic" ESP32 which has 3 HW UARTS.
EspSoftwareSerial can have issues with loopback due to limited full duplex capabilities inherent in CPU implemented SW UARTS, as far as I can tell, you are sending from Arduino `Serial` RX to EspSoftwareSerial TX to a third device's RX, and in the backward direction from that third device's TX to EspSoftwareSerial's RX and forward that to Arduino `Serial` TX? This works for everyone else!

dok-net, 2023-01-22:
No activity

closed 2023-01-22 by dok-net