plerup/espsoftwareserial #273: Possibility to make pin check methods public

x29a, 2023-02-17:
Hi there and thanks for this great library!

Is there a reason why the pin check methods (https://github.com/plerup/espsoftwareserial/blob/a97f09a3be5584499104b2e19668b5edccccbaa4/src/SoftwareSerial.h#L234)
``` 
    static constexpr bool isValidGPIOpin(int8_t pin);
    static constexpr bool isValidRxGPIOpin(int8_t pin);
    static constexpr bool isValidTxGPIOpin(int8_t pin);
```
are in the private section of the class declaration?

If they were public, one could use the library to find appropriate pins, thanks to ESP32 pin multiplexing, this could be done during runtime. If course, there needs to be a valid device connected still ;)

At least it makes it easier to print out "yes, the pins you have configured are correct for this library".

fanfanlatulipe26, 2023-02-18:
Have a look at :  add "weak" attribute to function isValidGPIO , isValidRxGPIO , isValidTxGPIO #251 
But is was closed
and also:   Use GPIO2 as TX #261

dok-net, 2023-02-18:
Implemented in https://github.com/plerup/espsoftwareserial/commit/d958f7b8ca53a2d896cae76848fcb314135586bc

closed 2023-02-18 by dok-net

x29a, 2023-02-18:
@fanfanlatulipe26 thanks for your pointers, i did look at the other open and closed issues for discussion but didnt find anything about accessibility of those methods, i think your weak linkage issue is more about the possibility to override the methods.

@dok-net thanks for the quick implementation!