plerup/espsoftwareserial #275: Higher bitrates possible for TX only?

shlomozippel, 2023-02-19:
I'm considering using this library for DMX output which has a baudrate of 250kbps. Should I even try to modify the library for that datarate (TX only)? Or is it a Bad Idea? Probably targeting ESP32-S3 only

dok-net, 2023-02-23:
Is this a use case where the HW UARTs cannot implement the wire protocol?
From a performance point of view, I have implemented TX in a fashion to recover as much idle time from the bit banging as possible, which may hurt performance, but is good enough at the usual bitrates for serial sensors etc. From the issues I have seen over the years, most run at 9600bps. I have never had success with over 150000bps or similar on classic ESP32s - without concurrent WiFi, even. So for TX only and a different protocol than UART (universal _asynchronous_ Rx/Tx) you should be able to wrap you own bit-banging routing in no time.
If you find different, that is, TX-only at 250000bps works with this lib, maybe give some detail on what makes DMX different from the TIA232 this library speaks, thanks in advance.

dok-net, 2023-02-25:
No WiFi, half-duplex, ESP8266, can do 250000bps max at 160 MHz CPU clock. But: IIRC, ESP32 is slower due to RTOS layer.

shlomozippel, 2023-02-25:
Thanks for looking into it.

I believe DMX uses standard TIA with specific settings (250kbps, 2 stop bits, 0 parity bits) with a special reset frame in between up-to-512 byte data frames. Those settings are pretty trivial for the hardware UART devices, I'm just looking for clever ways to get ~8 DMX outputs. Software bit banging is one option, I2S/LCD peripheral is another. I'm guessing it's also possible using RMT however there are only 4 RMT tx channels on ESP32-S3

Thanks again

shlomozippel, 2023-02-25:
Also I'm assuming that bit banging for 1 output vs. for 8 outputs will be similar if bit banging to the direct GPIO port register

dok-net, 2023-03-05:
@shlomozippel Honestly, I still don't understand that sentence after a week :-) Do you expect to send the same bit stream to a number of GPIO all at the same time? It increasingly sounds like this is nothing to do with EspSoftwareSerial that I am going to go into...

shlomozippel, 2023-03-05:
Happy to explain :)

When bit banging a protocol to a single pin you end up setting and clearing a GPIO with specific timing. It is also possible to set and clear multiple pins at a time by writing to a GPIO port instead of just a single pin. Each bit in the port register is mapped to a different GPIO pin so you can write to multiple pins by writing the correct mask to the SET and CLEAR port registers. Simple example to set 2 pins and clear 4 pins:

```GPIO_OUT_W1TS_REG = 0b0101; // set GPIO0 and GPIO2
GPIO_OUT_W1TC_REG = 0b1111; // clear GPIO0, GPIO1, GPIO2, GPIO3```

So if you have 8 stream of data and you transpose them into a single register which you write to `GPIO_OUT_W1TS_REG` then you effectively write out 8 parallel data streams on 8 different pins using a single instruction. So for EspSoftwareSerial that would mean output for 8 serial TX pins at the same time, which is my goal.

This is a pretty common method for parallel output. It's often used for parallel LED outputs like FastLED parallel output (https://github.com/FastLED/FastLED/wiki/Parallel-Output)

So, in other words, if it's possible to TX software serial to 1 pin at 250kbps then it's also possible to output to 8 of them in parallel by writing to the port register

dok-net, 2023-03-06:
@shlomozippel But it's not really possible to do 250kbps with the design chosen for EspSoftwareSerial. Feel free to experiment, of course. I'm closing this discussion now, because it's not about Software EIA232.

closed 2023-03-06 by dok-net