plerup/espsoftwareserial #276: Why do I need to delay in this loop？

shownb, 2023-02-22:
hello,
my code like that, I always have to delay a certain amount of time(>=800us) after myPort.read(), otherwise esp32 will restart.
with delayMicroseconds(800) it will be work perfect.
but if every byte need 800us, it will waste too much time.
```C++
#include <SoftwareSerial.h>

#define MYPORT_TX 12
#define MYPORT_RX 13

unsigned char bufDev[4096];
unsigned short lenDevData = 0;

SoftwareSerial myPort;

void setup() {
  myPort.begin(19200, SWSERIAL_8S1, MYPORT_RX, MYPORT_TX, false);
  myPort.enableIntTx(true);
  if (!myPort) {
    Serial.println("Invalid SoftwareSerial pin configuration, check config");
    while (1) { // Don't continue with invalid configuration
      delay (1000);
    }
  }
}

void loop() {
   read_from_dev()
}

void read_from_dev() {
  unsigned char *p = bufDev;
  while (myPort.available() > 0) {
    *p = myPort.read();
    lenDevData ++;
    if (lenDevData >= 4096) {
      break;
    }
    p ++;
    delayMicroseconds(800);
  }

  if (lenDevData > 0) {
//do something to clear 
lenDevData = 0;
  }
}

```

dok-net, 2023-02-23:
I doubt your issue is correct, if nothing else, because your edited sample wouldn't even compile, notice the missing semicolon:
```
void loop() {
   read_from_dev()
}
```

Therefore it can be assumed that whatever exactly `esp32 will restart` means, it happens in your code which you have omitted.
Could you please fix, build and run your exact MCVE and base your issue, if any, on that? Btw, what is an ESP32 :-) "Classic", -C3, -S2, -C6, et cetera. It's not as simple as it used to be, smiles :-)

shownb, 2023-02-23:
oh,you always this style. :(
here is the device chip_id
```bash
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
```

Eventually I found out what the problem was, if I didn't delay, then I wouldn't get the characters when they came later, they just came slower. I need to delay a bit and wait for them to come
```c++
  while (myPort.available() > 0) {
    *p = myPort.read();
//if we wait a little bit,myPort.available>0 is still true. then i can get the whole data.
    delayMicroseconds(800);
  }
```

please closed the issue.Once again, 
espsoftwareserial has proven to work well. Thank you for your work.

dok-net, 2023-02-23:
> oh,you always this style. :(

Maybe let's not get personal. I am surprise you knew you had it coming and still posted a community-support site type of question to this engineering issue tracking - obviously you did your homework but didn't respect my privacy :-)
Anyway, I see you were glad I did respond in short time, but I sometimes consider not caring anymore when there are too many issues of the kind of "I don't know precisely what I am doing, please help me" are stacking up here. See, I am resolved to seriously inspect every report with due diligence, so I am entitled to expect that reporters do the same, which wasn't the case in this issue.

> please closed the issue.

IIRC the OP can do this themselves.

closed 2023-02-23 by dok-net