plerup/espsoftwareserial #277: dangerous relocation: l32r: literal placed after use | When using BME280I2C library

38saibot, 2023-03-04:
Hi there,

I was using the Softwareserial in my temperature logger for some months and everything worked fine with version 6.17.1
Now I updated the lib in Platformio to 7.0.1.
Now I get this error when compiling:

`.pio\build\esp32dev\libee3\libEspSoftwareSerial.a(SoftwareSerial.cpp.o): in function `std::function<void ()>::operator()() const':
c:\users\tobi\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:686:(.iram1.66[_ZNK8delegate6detail12DelegateImplIPvvEclEv]+0x24): dangerous relocation: l32r: literal placed after use: .literal._ZNK8delegate6detail12DelegateImplIPvvEclEv
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1`

Now I found out after 2 hours of searching around, that the error only occurs, when I use the BME280 Library from here:
[https://github.com/finitespace/BME280](https://github.com/finitespace/BME280)

So I setup a minimalistic project with not more than this code:

`#include <Arduino.h>
#include <SoftwareSerial.h>
#include <BME280I2C.h>

BME280I2C bme;

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}`

Platformio setting:

`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	plerup/EspSoftwareSerial@^7.0.1
	finitespace/BME280@^3.0.0`

Unfortunaltey I have abolutely no glue what is going wrong here, I even cannot find the code line in SoftwareSerial where the problem occurs. Does anybody have an idea how I could solve this problem or which change in the SoftwareSerial since the mentioned version could cause this problem?

dok-net, 2023-03-05:
Oh, I am totally sorry I forget what was behind this the last time I noticed something similar. Perhaps an IRAM_ATTR misuse. Out of memory, sketch and libs are to large for the platform? As long as you have consistently updated all tools, cores, and libraries, I don't know. I think I am using a different BME280 lib, but I haven't build that project just yet. May do so later, perhaps. If you find your issue gets resolved by anything you do, please don't wait and report here ASAP, thank you.

pasleto, 2023-03-05:
I am getting exactly the same error 
`.pio\build\esp32dev\lib9c5\libespsoftwareserial.a(SoftwareSerial.cpp.o): in function `std::function<void ()>::operator()() const':
c:\users\tomas\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:686:(.iram1.66[_ZNK8delegate6detail12DelegateImplIPvvEclEv]+0x24): dangerous relocation: l32r: literal placed after use: .literal._ZNK8delegate6detail12DelegateImplIPvvEclEv
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1`
when using these libraries:

https://github.com/espressif/arduino-esp32/tree/master/libraries/DNSServer
`#include <DNSServer.h>`
https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer
`#include <WebServer.h>`
https://github.com/govorox/SSLClient
`#include <SSLClient.h>`

the same minimalistic setup as above was tested with each library, same scenario, even declaration triggers the error

38saibot, 2023-03-05:
Interesting and good to know that it's not only "my" problem.
I cannot imagine that it's out of memory. The sketch I posted above is only a few bytes. My whole project is much bigger and running without SoftwaSerial. It's a ESP32 DevKit C. If it runs out of RAM, mostly the compiler does not know this in advance, usually these are runtime errors.

pasleto, 2023-03-05:
Yep, same here, using this in so much bigger project, and it is running fine including SoftwareSerial on version 7.0.0

dok-net, 2023-03-06:
I am sorry, I don't know how to help, because this sketch:
```
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <BME280I2C.h>

BME280I2C bme;

void setup()
{
}

void loop()
{
}
```
builds fine (condensed build output):
```
Compiling debug version of 'issue277' for 'WEMOS D1 MINI ESP32'
elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o "...\issue277\espressif_d1_mini32\Debug\issue277.ino.bootloader.bin" "...Arduino\hardware\espressif\esp32\tools\sdk\esp32\bin\bootloader_dio_80m.elf"  ) ) 
esptool.py v4.2.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
esptool.py v4.2.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Program size: 277,789 bytes (used 21% of a 1,310,720 byte maximum) (170.63 secs)
Minimum Memory Usage: 22648 bytes (7% of a 327680 byte maximum) 
```
using these local git clones, HEAD of main at the time of writing:

[arduino-esp32](https://github.com/espressif/arduino-esp32/commit/0d84018d969309addacbcc3e3782c1fadc95fbc8)
[espsoftwareserial](https://github.com/plerup/espsoftwareserial/commit/fa23f027f7e6f29f63a273c0c14a7327b66c3bcf)
[finitespace/BME280](https://github.com/finitespace/BME280/commit/72a11d3b7d484995eb34c6bb8533c5ceb2aa4156)

Please double check, and if your errors persist, report at the same level of detail or better, unless I can reproduce there's little I can do.

pasleto, 2023-03-06:
Try to initialize SoftwareSerial instance aswell, for me it compile fine aswell in some cases when library is included, the problem comes up when you want to use it

dok-net, 2023-03-06:
@pasleto How do I say it politely :-) You are wasting my time... either you post an MCVE with all the evidence, or you don't report here. I have asked a hundred times, I'm AM taking this seriously, but it does only take away from interesting stuff, like the functionality you have been using successfully, if issues are hard or impossible to reproduce. It's not just me, every project works this way. For discussion among peers, please head over to some of the support forums elsewhere. No offense intended, really, please understand my perspective.

dok-net, 2023-03-06:
An If and a But... :-) This builds just fine, as well:
```
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <BME280I2C.h>

BME280I2C bme;

SoftwareSerial duh;

void setup()
{
	duh.begin(115200, SWSERIAL_5E1, -1, -1, false, 64, 0);
	duh.write(0, PARITY_EVEN);
}

void loop()
{
}
```
What else do you want me to try?

pasleto, 2023-03-06:
@dok-net Sorry for my quick reply from phone. Here is the full example with WebServer:

```
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <WiFi.h>
#include <WebServer.h>

const char* ssid = "........";
const char* password = "........";

WebServer server(80);

SoftwareSerial swSerial;

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }

  server.on("/", []() {
    server.send(200, "text/plain", "Hello World!");
  });

  server.begin();

  swSerial.begin(115200, SWSERIAL_5E1, -1, -1, false, 64, 0);
  if (swSerial.available()) swSerial.write(0, SWSERIAL_PARITY_EVEN);
}

void loop() {
  server.handleClient();
}
```

Which is working fine with version 7.0.0

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
lib_deps = plerup/EspSoftwareSerial@7.0.0
```

```
Compiling .pio\build\esp32dev\src\main.cpp.o
Linking .pio\build\esp32dev\firmware.elf
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.9% (used 22532 bytes from 327680 bytes)
Flash: [          ]   3.8% (used 249441 bytes from 6553600 bytes)
Building .pio\build\esp32dev\firmware.bin
esptool.py v4.4
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
```

But failing with version 7.0.1

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = default_16MB.csv
lib_deps = plerup/EspSoftwareSerial@7.0.1
```

```
.pio\build\esp32dev\libdd2\libEspSoftwareSerial.a(SoftwareSerial.cpp.o): in function `std::function<void ()>::operator()() const':
c:\users\tomas\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:686:(.iram1.66[_ZNK8delegate6detail12DelegateImplIPvvEclEv]+0x24): dangerous relocation: l32r: literal placed after use: .literal._ZNK8delegate6detail12DelegateImplIPvvEclEv
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

Even this minimal code is enough to trigger the error

```
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <WebServer.h>

WebServer server(80);

SoftwareSerial swSerial;

void setup() { }

void loop() { }
```

Used arduino-esp32 is version 2.0.6
Tested espsoftwareserial versions: 7.0.0 (working), 7.0.1 (failing), current master (failing)

38saibot, 2023-03-06:
The arduino-esp32 version that is supported in the latest platform-espressif32 6.0.1 is the 2.0.6
So maybe that's doing the trick if we have to use the 2.0.7 which ist not yet published in the platform.
I will check it later

dok-net, 2023-03-06:
@pasleto Your minimal webserver sketch final triggers the linker error on my build, too:
`std_function.h:686: (.iram1.66[_ZNK8delegate6detail12DelegateImplIPvvEclEv]+0x24)  dangerous relocation  l32r  literal placed after use  .literal._ZNK8delegate6detail12DelegateImplIPvvEclEv
`

Are you happy now (sarcastic smile)?

But fun aside, thanks, I can now try to apply a few changes. From what can be found online about this kind of error, I don't know if there is a permanent fix, or that this never would happen before the latest changes to EspSoftwareSerial.

38saibot, 2023-03-06:
@dok-net To be honest, yes I'm happy and hopeful that you can find the problem ;-) Thank you so much for your great work!

I tried it today, and my example sketch from above is compiling. I did not change anything. This makes me crazy.
However, my real project throws the error again.

So I went to the Arduino IDE (the old one) and tried different combinations. Arduino framework 2.0.6, 2.0.7 with espsoftwareserial 7.0.0 and 7.0.1 and all combinations worked in the arduino IDE.

But PlatformIO seems to make something different. The fact that my minimalistic sample is working today, but yesterday not, is weird. Maybe I'm doing anything wrong, or Platformio is doing anything I dont know. I'm not very familiar with all the details.

dok-net, 2023-03-07:
So we get this error (demangled):
```
SoftwareSerial.cpp.o: in function std::function<void ()>::operator()() const
Build failed for project 'issue277'
std_function.h:686: (.iram1.66[delegate::detail::DelegateImpl<void*, void>::operator()() const] 0x24)  dangerous relocation  l32r  literal placed after use  .literal.delegate::detail::DelegateImpl<void*, void>::operator()() const
```
I am afraid I have to ask you to escalate this to the ESP32 compiler team. IMHO, there is an issue with the way it does or does not place everything of std::function in IRAM that's required for it to work in ISR context.

I am stopping all exploration of this issue at this point.

dok-net, 2023-03-07:
@mcspr Could you please take a look at this? It doesn't happen with the ESP8266 builds, perhaps you do have any idea, or you could make the proper suggestion to the ESP32 devs? I expect this is not a blocker for release of 8.0.0 for ESP8266.

dok-net, 2023-03-08:
@38saibot @pasleto  Please test eaa3fb5650aac9f3b336b326ed081e795ac92f6a really hard, on my example it fixed the linker error.

38saibot, 2023-03-08:
@dok-net I tested it in different projects, in PlatformIO and Arduino IDE 1.8 and it finally worked without any errors.
So for me the problem seems to be solved with this version. Thanks a lot for your good work!

pasleto, 2023-03-09:
@dok-net I can also confirm that build passed for my project. Tested even on arduino-esp32 2.0.5 Thanks a lot!

dok-net, 2023-03-10:
Fixed in https://github.com/plerup/espsoftwareserial/releases/tag/8.0.1

closed 2023-03-10 by dok-net

selamanse, 2023-03-10:
had similar issues as described here. tag 8.0.1 compiles for esp32dev board.

pio Core 6.1.5·Home 3.4.3

dok-net, 2023-03-11:
@selamanse What are you trying to tell us in your last message?

selamanse, 2023-03-11:
> @selamanse What are you trying to tell us in your last message?

Just wanted to acknowledge that this fix works.