plerup/espsoftwareserial #278: Build Failed With Latest Version

SherifMEldeeb, 2023-03-05:
### **Enviroment Setup:**
* platform = espressif32
* board = esp32doit-devkit-v1
* framework = arduino

### **Issue Description:**
naming conflict with enum espSoftwareSerial::Config and esp32-hal-uart.h definitions.

### **Suggestion:**
Change naming convention for enum attributes.

dok-net, 2023-03-05:
Yes, you are right, unfortunately the ESP32 Arduino core uses the C preprocessor with C++ and this overreaches right into the EspSoftwareSerial sources. Effectively, the rename from SWSERIAL_8N1 to the succinct EspSoftwareSerial::Config::SERIAL_8N1 can't work - until PR https://github.com/espressif/arduino-esp32/pull/7926 gets merged. Please put your vote behind that, if you will.

dok-net, 2023-03-05:
With latest master, this issue should no longer exist. Please check and close if appropriate.

SherifMEldeeb, 2023-03-05:
> With latest master, this issue should no longer exist. Please check and close if appropriate.

Thanks for your quick response, Fixed and tested.

closed 2023-03-05 by SherifMEldeeb