plerup/espsoftwareserial #279: Issues with ""

Yammes, 2023-03-05:
Hi Forum

since longer time I tried to set up the 'SoftwareSerial' Libbrary..............
just today I saw, that there is a new version 8.0.0  and have the hope that this is working ;-))))
okay: maybe I'm the only one , where it is not working ;-)
in the last time I tried serveral Versions,: I have older ones (V528 saved in 2019, and 2weeks ago restarted struggeling with V700, because I need 8E1) always the same ....
the Result is: There is alwaays an Issue with something in "circular_queue":
an smal excerpt of the error logs: 
 - "\src/circular_queue/Delegate.h:42:55: error: expected initializer before 'vPtrToFunPtrExec'"
 - "src/circular_queue/Delegate.h:241:28: error: expected ';' at end of member declaration   "
 - "src/circular_queue/Delegate.h:241:62: error: ISO C++ forbids declaration of 'operator=' with no type [-fpermissive]"

does this seems to be an Issue here on my side? If yes, did some one have an Idea where to look?
I Think, that I have to insert missing ';' cant be true, because I cant imagine that there is such an 'simple' coding error ;-)) 
beswt regards Yam

dok-net, 2023-03-05:
It's unreleased as of this writing, only pushed to the master branch. Hm, about your issue - could you please edit the title to something descriptive? I wonder, are you building for either ESP8266 or one of the ESP32 chips? On both I am compiling locally, and I obviously don't have these compiler issues. Circular queue standalone even builds on many more targets, for years now. So, what is your toolchain?

Yammes, 2023-03-05:
Dear Dirk Kaar,
1st: at first Thanks for your Reply!! I appreciate this
2nd: how to title or name the Issue? honestly, I have now glue where exact the point is? 
     so, to point it out, is difficult to say ;-))
3rd: I'm struggeling since weeks (and spend lot of hours to figure out this Lib-Issue)
     so now was the time to ask (and wasting your time) sorry!!
4th: the Compiler errors shown in Arduino are very long (650 lines, see attached separate file)
5th: I'm 100% sure that this lib runs 'for years',
     there fore I noticed 'ironicly' "maybe I'm the only one...." ;-))

Thanks in advance for an hint, where I have to look.
best regards
Yam

dok-net, 2023-03-05:
Arduino 1.8.9? Arduino Core for ESP8266 version 2.5.0? Arduino Core for ESP32 1.0.2? And why are you updating EspSoftwareSerial among all that pretty outdated stuff? It's bound to fail in totally uninteresting ways, I'm sorry.
Please close this issue ASAP, update everything, don't try to update EspSoftwareSerial without checking the README (in short, for ESP8266 don't use the Arduino library mananger, it's supplied by the Core project).
If you use the Git projects, check all the READMEs, please. If you use the IDE's supplied dependencies, I would suggest you look there if something fails and I can't corroborate your report.
I am sure you will have a much better experience in the future if you keep everything matched, version-wise.

Yammes, 2023-03-05:
Dear Dirk Kaar,
"ja, Asche auf mein Haupt", as we say here.....
Thanks for Pointing out this. ;-)) 
As an excuse: due to the fact, that I working on two bigger projects
I'm affraid/carefull, maybe 'overcautious' with updates.....
and the "Rule": "never change a running system", is not only empty words for me, it's also a experience .....
not only in Arduino world ;-)) 
(i.e. a little thing was : after updating a JSON Lib from Ver5 ~~> Ver6: Result: nothing worked after that)
how ever, here in this case its maybe correct to blame me, for not updating some general SW Packages.....

Fortunately I have a 2nd Labtop, to test/check this complete new environement.......
and hopefully all the work(Software code) done until now, will running ;-))

All this comes up, because one little Modbus Energy-Meter needs as Serial '8E1' instead of 'usual' "8N1" ;-))
On ESP32 with HardwareSerial working, but not on ESP8266/Wemos D1 mini w/ SW-Serial, 
so I'm started looking for an 'new' SoftwareSerial.......

How ever, thanks for your review!!!!

Regards Yam

closed 2023-03-05 by Yammes

Yammes, 2023-03-05:
"Asche auf mein Haupt" 
I worked with my 'running' System since 2 Years w/o updating main SW IDE Environement.
The Rule: "_never change a running system_"  was in this case maybe not a good Idea

Thanks for your pointing this out!!!!

Regards Yam

Yammes, 2023-03-05:
P.S. my brain is still working... ;-) 
what I have "_at the back of my mind_" was an Info regarding 'new' Arduino IDE: 
(I just found it again, this was most probably the reason for not updating the system):
"_There are two versions of the Arduino IDE you can install: version 1 and version 2.
  At the moment, there are some plugins for the ESP32 (like the SPIFFS Filesystem Uploader Plugin) 
  that are **not yet supported** on Arduino 2.  So, if you intend to use the **SPIFFS** plugin 
  in the future, we recommend installing the legacy version 1.8.X_."
<-- due to the fact that I use the "**SPIFFS Upload tool**" , this was the reason to stay @ IDE Ver.1.8.9 ;-))

FYI: I use FS for JSON-File, JSON for Config-Values , Config-Values to avoid hardcoded Values , 
Config-Values used w/ WiFi-Manager-Parameters, Parameters to make the Project sharable and adaptive ... 
for me it is nice to adapt an Project with an Web-Interface, To leave this out, is not an option in my project....

I know that I don't know a lot, but this was most probably the reason for not updating the system. 

Sorry, wasting your time with my old System ;-))
and again: Thanks a lot for your effort !!
best regards
Yam