plerup/espsoftwareserial #280: Naming conflict of SoftwareSerial.h and  esp32-hal-uart.h

ntustzeus, 2023-03-06:
I create a new ESP32(esp32-s3-devkitc-1) project, use PlatformIO, search library EspSoftwareSerial  and install it. after building process, got following error message:

It seems like we have predefined SERIAL_5N1(esp32-hal-uart.h), and use the name in enum(SoftwareSerial.h), so compiler give us the syntax error message.

Is there a way for not changing the name of SERIAL_5N1 and others?
(let the SoftwareSerial.h not reference to other files?)

**esp32-hal-uart.h**
```
#define SERIAL_5N1 0x8000010
#define SERIAL_6N1 0x8000014
#define SERIAL_7N1 0x8000018
#define SERIAL_8N1 0x800001c
#define SERIAL_5N2 0x8000030
#define SERIAL_6N2 0x8000034
#define SERIAL_7N2 0x8000038
#define SERIAL_8N2 0x800003c
```

**SoftwareSerial.h**
`enum Config {
    SERIAL_5N1 = PARITY_NONE,
    SERIAL_6N1,
    SERIAL_7N1,
    SERIAL_8N1,
    SERIAL_5E1 = PARITY_EVEN,
    SERIAL_6E1,
    SERIAL_7E1,
    SERIAL_8E1,
    SERIAL_5O1 = PARITY_ODD,
    SERIAL_6O1,
    SERIAL_7O1,
    SERIAL_8O1,
    SERIAL_5M1 = PARITY_MARK,
...
}`

> C:/Users/b9015/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.h:28:20: error: expected identifier before numeric constant
>  #define SERIAL_5N1 0x8000010
>                     ^~~~~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:48:5: note: in expansion of macro 'SERIAL_5N1'
>      SERIAL_5N1 = PARITY_NONE,
>      ^~~~~~~~~~
> C:/Users/b9015/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.h:28:20: error: expected '}' before numeric constant      
>  #define SERIAL_5N1 0x8000010
>                     ^~~~~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:48:5: note: in expansion of macro 'SERIAL_5N1'
>      SERIAL_5N1 = PARITY_NONE,
>      ^~~~~~~~~~
> In file included from .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.cpp:23:
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:47:13: note: to match this '{'
>  enum Config {
>              ^
> In file included from C:/Users/b9015/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal.h:78,
>                  from C:/Users/b9015/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:36,
>                  from .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/circular_queue/circular_queue.h:24,
>                  from .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:25,
>                  from .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.cpp:23:
> C:/Users/b9015/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.h:28:20: error: expected unqualified-id before numeric constant
>  #define SERIAL_5N1 0x8000010
>                     ^~~~~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:48:5: note: in expansion of macro 'SERIAL_5N1'
>      SERIAL_5N1 = PARITY_NONE,
>      ^~~~~~~~~~
> In file included from .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.cpp:23:
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:96:38: error: expected class-name before ',' token
>  class UART : private GpioCapabilities, public Stream {
>                                       ^
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:116:31: error: 'Config' has not been declared
>      void begin(uint32_t baud, Config config,
>                                ^~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:119:31: error: 'Config' has not been declared
>      void begin(uint32_t baud, Config config,
>                                ^~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:123:31: error: 'Config' has not been declared
>      void begin(uint32_t baud, Config config,
>                                ^~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:127:31: error: 'Config' has not been declared
>      void begin(uint32_t baud, Config config = SERIAL_8N1) {
>                                ^~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:191:32: error: 'Parity' has not been declared
>      size_t write(uint8_t byte, Parity parity);
>                                 ^~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:196:54: error: 'Parity' has not been declared
>      size_t write(const uint8_t* buffer, size_t size, Parity parity);
>                                                       ^~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:197:51: error: 'Parity' has not been declared
>      size_t write(const char* buffer, size_t size, Parity parity) {
>                                                    ^~~~~~
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:288:5: error: 'Parity' does not name a type; did you mean 'Print'?
>      Parity m_parityMode;
>      ^~~~~~
>      Print
> In file included from .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.cpp:23:
> .pio/libdeps/esp32-s3-devkitc-1/EspSoftwareSerial/src/SoftwareSerial.h:316:1: error: expected declaration before '}' token
>  }; // namespace EspSoftwareSerial
>  ^
> *** [.pio\build\esp32-s3-devkitc-1\lib64c\EspSoftwareSerial\SoftwareSerial.cpp.o] Error 1

dok-net, 2023-03-06:
You are using an arbitrary revision from master; due to the way this project is developed, there are no feature branches but main, which sometimes may fail to perform on all platforms. That's what releases are for. That said, even on release of the upcoming 8.0.0 there may need to be some fixes required due to breaking changes that I may take the liberty to have, Parity was one of those. Now that you have apparently proven that this might be a foreseeable issue, I'll look into how it can be mitigated, thanks for reminding me.

dok-net, 2023-03-06:
Using the project configuration I have detailed in https://github.com/plerup/espsoftwareserial/issues/277, this sketch builds fine:
```
#include <SoftwareSerial.h>

SoftwareSerial duh;

void setup()
{
	duh.begin(115200, SWSERIAL_5E1, -1, -1, false, 64, 0);
	duh.write(0, PARITY_EVEN);
}

void loop()
{
}
```
No issue (?)

ntustzeus, 2023-03-06:
I download the code form Github Release and Main branch, SERIAL_5N1 has changed to SWSERIAL_5N1. (don't know why I got arbitrary revision when I search form platformIO)

now I got the same issue as #277. I tried your code above, same result.(use the API, both of Release and Main Branch)
I got the error even when comment out code in setup().

Maybe we need to check the development environment at first.

Visual Studio Code, PlatformIO (Core 6.1.7a3·Home 3.4.3)
Board: esp32-s3-devkitc-1 (it seems not in the support list in [README.md](https://github.com/plerup/espsoftwareserial#readme))

Please let me know if you need more info.

**platform.ini**
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
```

**Test Code:(ERROR)**
```
#include <SoftwareSerial.h>
SoftwareSerial duh;
void setup()
{
	// duh.begin(115200, SWSERIAL_5E1, -1, -1, false, 64, 0);
	// duh.write(0, PARITY_EVEN);
}

void loop()
{
}
```

> .pio\build\esp32-s3-devkitc-1\lib0aa\libSoftwareSerial.a(SoftwareSerial.cpp.o): in function `std::function<void ()>::operator()() const':
> c:\users\b9015\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:686:(.iram1.66[_ZNK8delegate6detail12DelegateImplIPvvEclEv]+0x24): dangerous relocation: l32r: literal placed after use: .literal._ZNK8delegate6detail12DelegateImplIPvvEclEv
> collect2.exe: error: ld returned 1 exit status
> *** [.pio\build\esp32-s3-devkitc-1\firmware.elf] Error 1

dok-net, 2023-03-06:
@ntustzeus But your comments don't belong to this already fixed issue, but in the best of cases it's additional information to https://github.com/plerup/espsoftwareserial/issues/277. It helps nobody if issues are hijacked.

closed 2023-03-06 by dok-net