plerup/espsoftwareserial #287: Would it be possible to integrate start bit timestamp recording in the main code line?

guido4096, 2023-04-28:
For a special usecase, code needs the exact timing of the startbit. I implemented this here: https://github.com/guido4096/espsoftwareserial/tree/add-startbit-timestamp

Is there a possibility to integrate this in the main repository? It adds minimal overhead, especially when not using the time stamp recording. In the critical bit crushing loop it adds one "assignment" and one "if statement" per byte received.

dok-net, 2023-04-29:
Hi. You may have felt that you would be overstepping by proposing a PR? No worries! It's way easier for me to review your net changes. Having said that, I have browsed into your repo's branch a little, and I don't think I would be willing to maintain such a little requested feature, down the road.
If I may make a suggestion: There exists the callback/notification for new data after buffer-empty now. Perhaps there is a way that you could exploit that as is? Or maybe just add an API to modify the behavior of that to suit your use-case on request? Either way, please issue a PR, mark it as draft, and I can at least view the state of your work here, online, without much trouble.

guido4096, 2023-04-29:
indeed, I sort of expected this change might not be in the general interest. I did try to make use of the existing callback, in fact still am, but the callback did not give me reliable start bit time, it calls back more randomly. I'll give it a thought if there is a nice way to add the behavior in a more on demand style. I consider this issue as resolved.

closed 2023-04-29 by guido4096