plerup/espsoftwareserial #288: incompatibility with arduino environment

gkwn1234400, 2023-05-17:
I have version 1.8.9 of arduino IDE installed on my computer. correspondingly for esp8266 the software version is "Arduino core for ESP8266 WiFi chip 3.0.2"
when I try to compile the loopback example, it returns the following error

loopback:50:11: error: 'EspSoftwareSerial' does not name a type; did you mean 'SoftwareSerial'?
   50 | constexpr EspSoftwareSerial::Config swSerialConfig = EspSoftwareSerial::SWSERIAL_8E1;
      |           ^~~~~~~~~~~~~~~~~
      |           SoftwareSerial
loopback:81:1: error: 'EspSoftwareSerial' does not name a type; did you mean 'SoftwareSerial'?
   81 | EspSoftwareSerial::UART serialIUT;
      | ^~~~~~~~~~~~~~~~~
      | SoftwareSerial
C:\Users\jeck2\Documents\Arduino\libraries\EspSoftwareSerial\examples\loopback\loopback.ino: In function 'void setup()':
loopback:110:5: error: 'serialIUT' was not declared in this scope
  110 |     serialIUT.begin(IUTBITRATE, swSerialConfig, D5, D6, invert, 2 * BLOCKSIZE);
      |     ^~~~~~~~~
loopback:110:33: error: 'swSerialConfig' was not declared in this scope; did you mean 'hwSerialConfig'?
  110 |     serialIUT.begin(IUTBITRATE, swSerialConfig, D5, D6, invert, 2 * BLOCKSIZE);
      |                                 ^~~~~~~~~~~~~~
      |                                 hwSerialConfig
C:\Users\jeck2\Documents\Arduino\libraries\EspSoftwareSerial\examples\loopback\loopback.ino: In function 'void loop()':
loopback:179:5: error: 'serialIUT' was not declared in this scope
  179 |     serialIUT.write(block, BLOCKSIZE);
      |     ^~~~~~~~~
loopback:212:19: error: 'swSerialConfig' was not declared in this scope; did you mean 'hwSerialConfig'?
  212 |         if (0 != (swSerialConfig & 070))
      |                   ^~~~~~~~~~~~~~
      |                   hwSerialConfig
loopback:219:23: error: 'swSerialConfig' was not declared in this scope; did you mean 'hwSerialConfig'?
  219 |             if (0 != (swSerialConfig & 070))
      |                       ^~~~~~~~~~~~~~
      |                       hwSerialConfig
loopback:225:58: error: 'swSerialConfig' was not declared in this scope; did you mean 'hwSerialConfig'?
  225 |                 expected = (expected + 1) % (1UL << (5 + swSerialConfig % 4));
      |                                                          ^~~~~~~~~~~~~~
      |                                                          hwSerialConfig
loopback:232:62: error: 'swSerialConfig' was not declared in this scope; did you mean 'hwSerialConfig'?
  232 |             if (serialIUT.readParity() != (static_cast<bool>(swSerialConfig & 010) ? serialIUT.parityOdd(r) : serialIUT.parityEven(r)))
      |                                                              ^~~~~~~~~~~~~~
      |                                                              hwSerialConfig
loopback:254:33: error: 'swSerialConfig' was not declared in this scope; did you mean 'hwSerialConfig'?
  254 |         uint8_t wordBits = (5 + swSerialConfig % 4) + static_cast<bool>(swSerialConfig & 070) + 1 + ((swSerialConfig & 0300) ? 1 : 0);
      |                                 ^~~~~~~~~~~~~~
      |                                 hwSerialConfig
Multiple libraries were found for "SoftwareSerial.h"
 Used: C:\Users\jeck2\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\SoftwareSerial
 Not used: C:\Users\jeck2\Documents\Arduino\libraries\EspSoftwareSerial
exit status 1
'EspSoftwareSerial' does not name a type; did you mean 'SoftwareSerial'?

dok-net, 2023-05-18:
That's quite old, plus I can safely assume than you haven't read the documentation first before asking for help. So, I repeat myself: don't mix versions of the core and EspSoftwareSerial, that's naturally including the examples, too.

closed 2023-05-18 by dok-net

gkwn1234400, 2023-05-19:
I'm sorry for your time .
I m  but I m beginner in esp so is difficult for me to use esp without core
.

In fact I don't now how to use esp8266 without core .