plerup/espsoftwareserial #291: Compile erro with updated Arduino Framework and esp tool

softwarecrash, 2023-06-17:
Hello,
i have some trubble with softwareserial,
on my one pc with Visual Studio Code and PlatformIO it work like a charme, and on another pc it trow a lot of error.
so i found only the difference in the Packeges
working with:
```PLATFORM: Espressif 8266 (4.1.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30101.0 (3.1.1) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)```

and with 
```PLATFORM: Espressif 8266 (2023.3.0) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 2.7.4+9 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa @ 2.40802.200502 (4.8.2)```
 
it throws this errors:
```.pio\libdeps\d1_mini_debug\PubSubClient\src\PubSubClient.cpp: In member function 'boolean PubSubClient::publish_P(const char*, const uint8_t*, unsigned int, boolean)':
.pio\libdeps\d1_mini_debug\PubSubClient\src\PubSubClient.cpp:523:19: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     return (rc == expectedLength);
                   ^
In file included from src\PI_Serial\PI_Serial.h:1:0,
                 from src\PI_Serial\PI_Serial.cpp:3:
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h: In static member function 'static constexpr bool EspSoftwareSerial::GpioCapabilities::hasPullUp(int8_t)':
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:102:5: error: body of constexpr function 'static constexpr bool EspSoftwareSerial::GpioCapabilities::hasPullUp(int8_t)' not a return-statement
     }
     ^
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h: At global scope:
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:285:23: warning: 'deprecated' attribute directive ignored [-Wattributes]
     void perform_work();
                       ^
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:199:9: error: 'int EspSoftwareSerial::UARTBase::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:226:9: error: 'int EspSoftwareSerial::UARTBase::read(uint8_t*, size_t)' marked override, but does not override
     int read(uint8_t* buffer, size_t size)
         ^
*** [.pio\build\d1_mini_debug\src\PI_Serial\PI_Serial.cpp.o] Error 1
In file included from src\PI_Serial/PI_Serial.h:1:0,
                 from src\main.cpp:24:
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h: In static member function 'static constexpr bool EspSoftwareSerial::GpioCapabilities::hasPullUp(int8_t)':
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:102:5: error: body of constexpr function 'static constexpr bool EspSoftwareSerial::GpioCapabilities::hasPullUp(int8_t)' not a return-statement
     }
     ^
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h: At global scope:
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:285:23: warning: 'deprecated' attribute directive ignored [-Wattributes]
     void perform_work();
                       ^
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:199:9: error: 'int EspSoftwareSerial::UARTBase::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
.pio\libdeps\d1_mini_debug\EspSoftwareSerial\src/SoftwareSerial.h:226:9: error: 'int EspSoftwareSerial::UARTBase::read(uint8_t*, size_t)' marked override, but does not override
     int read(uint8_t* buffer, size_t size)
         ^
*** [.pio\build\d1_mini_debug\src\main.cpp.o] Error 1```

dok-net, 2023-06-18:
This looks like the usual problem of PEBCAK: RTFM, don't use any other version from any library manager but the one that is intrinsically supplied by the core itself. This question comes up too frequently by far. @mcspr Perhaps we should remove EspSoftwareSerial from the core? I am considering splitting off the queue code into an own package, anyway, so something soon something has to change in any case....

closed 2023-06-18 by dok-net

mcspr, 2023-06-19:
Not sure how that would solve the build issue above, when old Core & toolchain would break with a mainline build at some point :shrug:

PIO usually gives a hint of where the library is coming from when running in verbose mode
> Dependency Graph
|-- EspSoftwareSerial @ 8.0.0+sha.b9a0f6e (License: LGPL-2.1+, URI: git+https://github.com/dok-net/espsoftwareserial#b9a0f6e0a2e95a1f7632edd9ce46d229ca1698ef, Path: /home/runner/swserial277/.pio/libdeps/d1_mini/EspSoftwareSerial)

And project file has to enable it in the first place
```ini
[env]
framework = arduino
lib_deps =
    https://github.com/dok-net/espsoftwareserial#b9a0f6e0a2e95a1f7632edd9ce46d229ca1698ef

[env:d1_mini]
platform = espressif8266
board = d1_mini
```