plerup/espsoftwareserial #293: Can't build for ESP32 in Platform.io: build error

DeflateAwning, 2023-06-22:
Version: `plerup/EspSoftwareSerial@^8.0.3`

Error output:
```
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\Stream.cpp.o
In file included from .pio\libdeps\esp32doit-devkit-v1\EspSoftwareSerial\src\circular_queue/circular_queue.h:31:0,
                 from .pio\libdeps\esp32doit-devkit-v1\EspSoftwareSerial\src\SoftwareSerial.h:25,
                 from .pio\libdeps\esp32doit-devkit-v1\EspSoftwareSerial\src\SoftwareSerial.cpp:23:
.pio\libdeps\esp32doit-devkit-v1\EspSoftwareSerial\src\circular_queue/Delegate.h: In member function 'R delegate::detail::DelegateImpl<A, R>::operator()() const [with A = void*; R = void]':
.pio\libdeps\esp32doit-devkit-v1\EspSoftwareSerial\src\circular_queue/Delegate.h:1170:25: error: R delegate::detail::DelegateImpl<A, R>::operator()() const [with A = void*; R = void] causes a section type conflict with void EspSoftwareSerial::UARTBase::preciseDelay()
             R IRAM_ATTR operator()() const
                         ^
.pio\libdeps\esp32doit-devkit-v1\EspSoftwareSerial\src\SoftwareSerial.cpp:287:16: note: 'void EspSoftwareSerial::UARTBase::preciseDelay()' was declared here
 void IRAM_ATTR UARTBase::preciseDelay() {
                ^
Compiling .pio\build\esp32doit-devkit-v1\FrameworkArduino\StreamString.cpp.o
*** [.pio\build\esp32doit-devkit-v1\lib2e2\EspSoftwareSerial\SoftwareSerial.cpp.o] Error 1
```

Any idea what's happening?

dok-net, 2023-06-24:
Are you setting `-Werror`? Could you try `-Wno-error=attributes`?

DeflateAwning, 2023-06-24:
Just using the default Platform.io installation in VS Code. Let me know if you can reproduce with one of the example codes.

dok-net, 2023-06-25:
> Just using the default Platform.io installation in VS Code. Let me know if you can reproduce with one of the example codes.

I repeatedly say that I don't support Platform.io. As long as there are no issues with ESP8266 Arrduio Core and the contained EspSoftwareSerial version, and also happily ESP32 Arduino Core master branch plus EspSoftwareSerial master branch, thiings are without flaw per definiition.
What's wrong with my suggested fix, have you tried it? In Arduino, please learn about `platform.local.txt` to modify compiler and linker flags. I am watching the developing situation with the GCC version upgrade in ESP32 IDF and Arduino core. The rule of thumb is, that more often than not new EspSoftwareSerial releases predate releases of the Arduino cores and rely on the master branch. ESP8266 has that dependency nailed, ESP32 is a little more tricky, but then again, it's not such a major moving target most of the time that it's asking too much of users. Platformio appears a little difficult in that regard, despite their much more detailed dependency resolution. If someone would propose a PR to improve the metadata file for Platformio here, properly detailing the issue and then some positive feedback from other users, I am more than willing to merge such things. But, again, I don't know, it's outside my volunteer support :-)
Please consider if there can be any progress on this issue, otherwise perhaps close it as this is the wrong place to ask?

cuspis, 2023-06-26:
[image]

DeflateAwning, 2023-06-26:
It would be awesome if you supported PlatformIO, but I understand if it's too much. A big warning in the README might be valuable if it doesn't work in PIO though, (because currently there doesn't appear to be a mention).

Where is this platform.local.txt file?

dok-net, 2023-07-02:
"It doesn't work in PIO" - I would like to contradict this generalization.
"Where is this platform.local.txt" - Please don't ask for every single bit of information here, there's so much more to learn and have fun with if you research these things online by yourself. In case I haven't made myself clear previously, the platform.local.txt is for Arduino, I don't know that PIO honors any of that or uses a different strategy.

Either way, I'm rebuilding a few of the tests and if everything looks good to me, I will be releasing another version.

dok-net, 2023-07-02:
@cuspis Please don't post screenshots the clip the interesting bits, but at least copy&paste the full text.

dok-net, 2023-08-05:
This issues appears to have gone stale.

closed 2023-08-05 by dok-net

DeflateAwning, 2023-08-07:
Hmm closing an unsolved issue seems like bad practice