plerup/espsoftwareserial #297: difference between baudrate setting and read back

hermanius, 2023-09-29:
Hi,

I'm on the ESP32-C3-DevkitM-1

when I set the baudrate with .begin(...) to 9600, and read it back with .baudRate() it shows 9615 baud.
When I do the same with 115200, .baudRate() gives back 125000.

When I communicate at 9600 it seems to work (most of the time), but 115200 communication does not work.
Why is there a difference between the set baudrate with .begin(...) and .baudRate(), and can it cause problems with the communication between the C3 and the peripheral device?

dok-net, 2023-10-07:
Currently, the true datarate is based internally on the precision of `micros()`, and reported by `baudRate()`. IIRC the switch from CPU cycles to micros() was done to mitigate CPU frequency scaling issues that get introduced by concurrent use of the WiFi stack. This was, to be honest, not completely tested, but the basic assumption is a lesson learned from the implementation of the current ESP8266 SW PWM generator.
I will experiment with returning to CPU cycles, as a preprocessing option. But I cannot commit to any due day. In the meantime.

As for 115200bps on anything other than ESP8266, I think it never predictably worked at that high bitrate.

dok-net, 2023-10-08:
See issue #260 . So, while reverting to CPU cycles provides higher bitrates at useable error rate on the basic ESP32, it was seen as non-viable on the later ESP32 variants mentioned in the PR. I will analyse a compromise solution, but it will take time.
I am not a fan of nagging open issues in this project, that I am aware of, and may/may not get fixed at an unspecified date in the future, therefore I'm closing the issue.

closed 2023-10-08 by dok-net