plerup/espsoftwareserial #305: fatal error: circular_queue.h: No such file or directory

red-scorp, 2024-01-07:
I've noticed following error during building with PlatformIO:

In file included from ag-panel/src/uart/SoftwareUART.cpp:10:
.piolibdeps/lolin_c3_mini/EspSoftwareSerial/src/SoftwareSerial.h:24:10: fatal error: circular_queue.h: No such file or directory

Following configuration are failing this way:

;   WeMos D1 Mini and clones based on Esp32 chip
[env:wemos_d1_mini32]
platform = espressif32
board = wemos_d1_mini32
lib_deps = ${common.lib_deps}
    erropix/ESP32 AnalogWrite
    plerup/EspSoftwareSerial

;   WeMos LOLIN C3 mini
[env:lolin_c3_mini]
platform = espressif32
board = lolin_c3_mini
lib_deps = ${common.lib_deps}
    erropix/ESP32 AnalogWrite
    plerup/EspSoftwareSerial

;   WeMos LOLIN S3 mini
[env:lolin_s3_mini]
platform = espressif32
board = lolin_s3_mini
lib_deps = ${common.lib_deps}
    erropix/ESP32 AnalogWrite
    plerup/EspSoftwareSerial

wtfprogrammer, 2024-01-07:
Looks like the dependency is broken:
Library Manager: Warning! Could not install `{'owner': 'dok-net', 'name': 'ghostl', 'version': '^1.0.0'}` dependency for the`EspSoftwareSerial` package

Falling back to 7.0.1 helps.

happen-studio, 2024-01-07:
even with 7.0.1 I get a lot of errors:

`In file included from .pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.cpp:23:0:
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:231:23: warning: 'deprecated' attribute directive ignored [-Wattributes]
     void perform_work();
                       ^
In file included from src/main.cpp:2:0:
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:231:23: warning: 'deprecated' attribute directive ignored [-Wattributes]
     void perform_work();
                       ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:145:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:172:9: error: 'int SoftwareSerial::read(uint8_t*, size_t)' marked override, but does not override
     int read(uint8_t* buffer, size_t size)
         ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:145:9: error: 'int SoftwareSerial::availableForWrite()' marked override, but does not override
     int availableForWrite() override {
         ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:172:9: error: 'int SoftwareSerial::read(uint8_t*, size_t)' marked override, but does not override
     int read(uint8_t* buffer, size_t size)
         ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.cpp: In static member function 'static constexpr bool SoftwareSerialGpioCapabilities::hasPullUp(int8_t)':
.pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.cpp:85:1: error: body of constexpr function 'static constexpr bool SoftwareSerialGpioCapabilities::hasPullUp(int8_t)' not a return-statement
 }
 ^
In file included from .pio/libdeps/nodemcu/EspSoftwareSerial/src/circular_queue/circular_queue.h:31:0,
                 from .pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:27,
                 from .pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.cpp:23:
.pio/libdeps/nodemcu/EspSoftwareSerial/src/circular_queue/Delegate.h: At global scope:
.pio/libdeps/nodemcu/EspSoftwareSerial/src/circular_queue/Delegate.h:1057:23: error: section of 'delegate::detail::DelegateImpl<A, R>::operator bool() const [with A = void*; R = void]' conflicts with previous declaration
             IRAM_ATTR operator bool() const
                       ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/circular_queue/Delegate.h:1126:25: error: section of 'R delegate::detail::DelegateImpl<A, R>::operator()() const [with A = void*; R = void]' conflicts with previous declaration
             R IRAM_ATTR operator()() const
                         ^
In file included from .pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.h:27:0,
                 from .pio/libdeps/nodemcu/EspSoftwareSerial/src/SoftwareSerial.cpp:23:
.pio/libdeps/nodemcu/EspSoftwareSerial/src/circular_queue/circular_queue.h:111:22: error: section of 'size_t circular_queue<T, ForEachArg>::available() const [with T = unsigned int; ForEachArg = SoftwareSerial*; size_t = unsigned int]' conflicts with previous declaration
     size_t IRAM_ATTR available() const
                      ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/circular_queue/circular_queue.h:175:20: error: section of 'bool circular_queue<T, ForEachArg>::push(T&&) [with T = unsigned int; ForEachArg = SoftwareSerial*]' conflicts with previous declaration
     bool IRAM_ATTR push(T&& val)
                    ^
.pio/libdeps/nodemcu/EspSoftwareSerial/src/circular_queue/circular_queue.h:198:20: error: section of 'bool circular_queue<T, ForEachArg>::push(const T&) [with T = unsigned int; ForEachArg = SoftwareSerial*]' conflicts with previous declaration
     bool IRAM_ATTR push(const T& val)
                    ^
*** [.pio/build/nodemcu/src/main.cpp.o] Error 1
*** [.pio/build/nodemcu/libae8/EspSoftwareSerial/SoftwareSerial.cpp.o] Error 1`

happen-studio, 2024-01-07:
it works if I downgrade the platform version

tbarbette, 2024-01-07:
Just installed esp32 from scratch in VSCode with PlatformIO and got here too. I'm a noob here, does anyone has a quickfix?

ashb, 2024-01-07:
The solution for now until this ghostl is relreased properly seems to be to add this to the `lib_deps` section:

```ini
  https://github.com/dok-net/ghostl
```

i.e.

```ini
lib_deps =
  ${env.lib_deps}
  plerup/EspSoftwareSerial
  https://github.com/dok-net/ghostl
```

dok-net, 2024-01-09:
Could not one of all the platformio users please issue a request to include ghostl in the online library manager? From what I read, it's as simple as command line oneline statement. I don't think there's any responsibility involved, perhaps beyond getting it right this one first time.

dok-net, 2024-01-09:
And perhaps rather continue this discussion [here](https://github.com/dok-net/ghostl/issues/1) instead.

closed 2024-01-09 by dok-net

red-scorp, 2024-01-09:
> ```ini
> lib_deps =
>   ${env.lib_deps}
>   plerup/EspSoftwareSerial
>   https://github.com/dok-net/ghostl
> ```

This works. Thanks!
Adding ghostl properly to PlatformIO library registry is also good idea!