plerup/espsoftwareserial #307: Ghostl library not installed via PlatformIO

CraigBurden, 2024-01-11:
Hi there, 

When adding this library to a PIO project using `plerup/EspSoftwareSerial@^8.2.0` the build fails as the `ghostl` library has not been installed and the `circular_queue.h` file is not found. 

This can easily be fixed by adding `https://github.com/dok-net/ghostl` to lib_deps, but should not be required if the dependency is setup correctly

sartyx, 2024-01-13:
doesn't really work? 

```
Library Manager: Installing dok-net/ghostl @ ^1.0.0
Library Manager: Warning! Could not install `{'owner': 'dok-net', 'name': 'ghostl', 'version': '^1.0.0'}` dependency for the`EspSoftwareSerial` package

```

dok-net, 2024-01-14:
@sartyx @CraigBurden @sjorsa I've spoken to the original EspSoftwareSerial maintainer, from whom I've taken over. He wasn't the one who registered EspSoftwareSerial with platform.io either - but isn't able to find out who it was and who's potentially behind registering new releases.
And no, it's not necessary for the registrant to the same person as the development lead. This is open source. If nobody takes any perceived "risk" to have a little effort, nobody is entitled to complain, either :-)
If you have an platformio account or are willing to create one, and if you can think you can show due dilligence in submitting, reviewing the result and perhaps updating for new releases in the forseeable near future, please go ahead.

popos123, 2024-02-24:
Hi, after installing git basch for windows, I was try to add lib_deps =
`maxgerhardt/ghostl@^1.0.0`
and
`https://github.com/dok-net/ghostl/archive/refs/tags/1.0.0.zip`
and
`https://github.com/dok-net/ghostl/`

and still **doesn't work**, ive got an error:
```
Resolving pico32 dependencies...
Library Manager: Installing dok-net/ghostl @ ^1.0.0
UnknownPackageError: Could not find the package with 'dok-net/ghostl @ ^1.0.0' requirements for your system 'windows_amd64'
```

verilog-indeed, 2024-03-02:
> doesn't really work?
> 
> ```
> Library Manager: Installing dok-net/ghostl @ ^1.0.0
> Library Manager: Warning! Could not install `{'owner': 'dok-net', 'name': 'ghostl', 'version': '^1.0.0'}` dependency for the`EspSoftwareSerial` package
> ```

Just had the same issue, solved it by cloning the `https://github.com/dok-net/ghostl` repo into my `libs` folder, make sure that dok-net/ghostl is NOT part of lib_deps.

Bascy, 2024-03-12:
Solved it bij splitting name and version in library.json

```
"dependencies": {
        "name": "ghostl",
        "version": "maxgerhardt/ghostl@^1.0.0"
    }
```
You can also specify a direct url into the github repo in stead of `<owner>/<repo>` construction

popos123, 2024-03-27:
so anyone cane make a new release e.g. 8.3.0 with two critical fixes from this #307 and #311 and optionally from #310?
really please, this library is almost perfect :(

dok-net, 2024-04-21:
Fixed by publishing the ghostl library to PIO under my own account now.