plerup/espsoftwareserial #309: error: atomic: No such file or directory

Djeremaille, 2024-02-16:
Hi 
when compiling I have this error.
I had many successful compile before and I touched nothing.

```
In file included from /home/minipc/Arduino/libraries/EspSoftwareSerial/src/SoftwareSerial.h:25,
                 from /media/nfsnas/Documents/Electronique/Interphone/sketch_dec18a/sketch_dec18a.c:1:
/home/minipc/Arduino/libraries/EspSoftwareSerial/src/circular_queue/circular_queue.h:28:10: fatal error: atomic: No such file or directory
 #include <atomic>
          ^~~~~~~~
compilation terminated.

Used library        Version Path
EspSoftwareSerial   8.1.0   /home/minipc/Arduino/libraries/EspSoftwareSerial
InputDebounce       1.6.0   /home/minipc/Arduino/libraries/InputDebounce
WiFi                2.0.0   /home/minipc/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/WiFi
AsyncTCP            1.1.4   /home/minipc/Arduino/libraries/AsyncTCP
ESP Async WebServer 2.7.0   /home/minipc/Arduino/libraries/ESP_Async_WebServer
FS                  2.0.0   /home/minipc/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/FS
WebSerial           1.4.0   /home/minipc/Arduino/libraries/WebSerial

Used platform Version Path
esp32:esp32   2.0.14  /home/minipc/.arduino15/packages/esp32/hardware/esp32/2.0.14
Error during build: exit status 1
```

svdrummer, 2024-02-16:
#include <atomic.h>

svdrummer, 2024-02-17:
And?

Djeremaille, 2024-02-18:
> And?

I tested but that doesn't work:

```
❯ arduino-cli compile --fqbn esp32:esp32:esp32 sketch_dec18a.ino
In file included from /home/minipc/Arduino/libraries/EspSoftwareSerial/src/SoftwareSerial.h:25,
                 from /media/nfsnas/Documents/Electronique/Interphone/sketch_dec18a/sketch_dec18a.ino:1:
/home/minipc/Arduino/libraries/EspSoftwareSerial/src/circular_queue/circular_queue.h:28:10: fatal error: atomic.h: No such file or directory
 #include <atomic.h>
          ^~~~~~~~~~
compilation terminated.

Used library      Version Path
EspSoftwareSerial 8.1.0   /home/minipc/Arduino/libraries/EspSoftwareSerial

Used platform Version Path
esp32:esp32   2.0.14  /home/minipc/.arduino15/packages/esp32/hardware/esp32/2.0.14
Error during build: exit status 1
```

svdrummer, 2024-02-18:
The compiler is telling us that that library  file is not there.  Reload the library or check the name.