plerup/espsoftwareserial #314: circular_queue.h: No such file or directory

naufalrizri, 2024-03-21:
im working with software serial and i encounter this issue

fatal error: circular_queue.h: No such file or directory
 #include <circular_queue.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.

popos123, 2024-03-27:
maybe check all files, or older release, or from fork that i was writing about this in another issue, maybe try using quotes instead <>, simple error, please check all possible solutions

jenschr, 2024-04-07:
I can confirm. This will happen to anyone trying to use the library currently. It is referenced here as well https://community.platformio.org/t/fatal-error-circular-queue-h-no-such-file-or-directory-espsoftwareserial-h/37959

This is a 100% breaking change since it is impossible to use the library without the file.

jenschr, 2024-04-07:
Adding the files manually from version 7.0.2 also does not work since the library will then throw lots of new errors such as this:

SoftwareSerial.h:328:44: error: expected ';' at end of member declaration
     static inline uint32_t IRAM_ATTR ticks() ALWAYS_INLINE_ATTR {
                                            ^
                                             ;

rtek1000, 2025-01-07:
I found this other library, I'm still going to test it, but it didn't give this initial error (fatal error: circular_queue.h: No such file or directory):

https://github.com/junhuanchen/Esp32-SoftwareSerial