plerup/espsoftwareserial #316: Esp32 S3

Flouzl, 2024-04-01:
hi,
could this library support Esp32 S3 ?

ihayri, 2024-04-05:
I would love to be able to use it with the S3 as well, I need 4 serial connections and am stuck.

doug-foster, 2025-05-05:
Any update on this?

Or could you suggest an alternative? Trying ESPSoftwareSerial 8.1.0 in Arduio IDE 2.3.6 on a SparkFun Thing Plus ESP32-S3. Get the linker error below.

Thanks!

Linking everything together...
/Users/dougfoster/Library/Arduino15/packages/esp32/tools/esp-x32/2411/bin/xtensa-esp32s3-elf-g++ -Wl,--Map=/Users/dougfoster/Library/Caches/arduino/sketches/7B6E208460B96D5BC399FA92035B362D/debug.ino.map -L/Users/dougfoster/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-2f7dcd86-v1/esp32s3/lib -L/Users/dougfoster/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-2f7dcd86-v1/esp32s3/ld -L/Users/dougfoster/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-2f7dcd86-v1/esp32s3/qio_qspi -Wl,--wrap=esp_panic_handler @/Users/dougfoster/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-2f7dcd86-v1/esp32s3/flags/ld_flags @/Users/dougfoster/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-2f7dcd86-v1/esp32s3/flags/ld_scripts -Wl,--start-group /Users/dougfoster/Library/Caches/arduino/sketches/7B6E208460B96D5BC399FA92035B362D/sketch/debug.ino.cpp.o /Users/dougfoster/Library/Caches/arduino/sketches/7B6E208460B96D5BC399FA92035B362D/libraries/EspSoftwareSerial/SoftwareSerial.cpp.o /Users/dougfoster/Library/Caches/arduino/cores/2b3883c2591773061dc4e23e61e5abf7/core.a @/Users/dougfoster/Library/Arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-2f7dcd86-v1/esp32s3/flags/ld_libs -Wl,--end-group -Wl,-EL -o /Users/dougfoster/Library/Caches/arduino/sketches/7B6E208460B96D5BC399FA92035B362D/debug.ino.elf
/Users/dougfoster/Library/Caches/arduino/sketches/7B6E208460B96D5BC399FA92035B362D/libraries/EspSoftwareSerial/SoftwareSerial.cpp.o: in function `_ZNK8delegate6detail12DelegateImplIPvvEclEv':
/Users/dougfoster/Library/CloudStorage/Dropbox/Data/doug/Topics/_dev-arduino/IDE - Arduino/libraries/EspSoftwareSerial/src/circular_queue/Delegate.h:1189:(.iram1.11[_ZNK8delegate6detail12DelegateImplIPvvEclEv]+0x2e): dangerous relocation: l32r: literal placed after use: .literal._ZNK8delegate6detail12DelegateImplIPvvEclEv
/Users/dougfoster/Library/Caches/arduino/sketches/7B6E208460B96D5BC399FA92035B362D/libraries/EspSoftwareSerial/SoftwareSerial.cpp.o: in function `_ZNK8delegate6detail13DelegatePImplIPN17EspSoftwareSerial8UARTBaseEvJOmEEclES5_':
/Users/dougfoster/Library/CloudStorage/Dropbox/Data/doug/Topics/_dev-arduino/IDE - Arduino/libraries/EspSoftwareSerial/src/circular_queue/Delegate.h:353:(.iram1.4[_ZNK8delegate6detail13DelegatePImplIPN17EspSoftwareSerial8UARTBaseEvJOmEEclES5_]+0x36): dangerous relocation: l32r: literal placed after use: .literal._ZNK8delegate6detail13DelegatePImplIPN17EspSoftwareSerial8UARTBaseEvJOmEEclES5_
collect2: error: ld returned 1 exit status
Using library EspSoftwareSerial at version 8.1.0 in folder: /Users/dougfoster/Library/CloudStorage/Dropbox/Data/doug/Topics/_dev-arduino/IDE - Arduino/libraries/EspSoftwareSerial 
exit status 1

Compilation error: exit status 1