plerup/espsoftwareserial #323: Fatal error: atomic: No such file or directory

AlbertHoevenaars, 2024-09-29:
Installed the latest ESPSoftware Serial library in the Arduino IDE Version: 2.3.3.

When I compile my software I got an error on the line "#include <SoftwareSerial.h>"
[image]

Is there a solution for the problem?

Thanks in advance

mjsneijders, 2024-09-30:
while trying to read in to this library i notice your isse,   and it seems to me,  that you have your libraries mixed up....  the error you get comes from a header file in a folder circular_queue,  and the library in this github does not have such a subfolder/header file...
you might want to look into that.

AlbertHoevenaars, 2024-10-01:
mjsneijders, thanks for the repley. I removed in the Arduino IDE the ESPSoftwareSerial libraries. Checked with the Windows explorer that the directory is gone. Then I installed the library ESPSoftwareSerial from Dirk Laar. In the attached picture you can see in the installed library there is a directory called "Circulair_queue".
When I look in the Github repostory at "https://github.com/plerup/espsoftwareserial/tree/main/src" then I don't see a circulair_queue directory. 
In the Softwareserial.h file there is still the line "#include <circular_queue.h>" and that *.h file is not found.
Don't understand what is going wrong.
[image]

AlbertHoevenaars, 2024-10-01:
When I run one of the example ino file I get the same error. "circular_queue.h No such file or directory"

mjsneijders, 2024-10-01:
must be some incompatibility somewhere..     it compiles for me,   what i did:
i installed latest arduino environment from the website... 2.3.3
i added the Esp32 boards from espressif.    ( by adding https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json  to the additional board managers url in the arduino preferences,   and then adding espressif esp32 boards in the board manager. 
i added esp32SoftwareSerial v 8.10 using the arduino library manager.  
opened an example (onewiretest) from the EspSoftwareSerial examples and clicked "Verify" 

this resulted in no errors.

mjsneijders, 2024-10-01:
[image]

AlbertHoevenaars, 2024-10-02:
After uninstalling and reinstalling the Arduino IDE I added the board manager URLs in the Preferences. Added the library ESPSoftwareSerial version 8.1.0. Opened the onewiretest.ino and compiling was successful.  Don't know what was wrong, but its working now.
mjsneijders thanks for the help!